zephyrproject-rtos/sdk-ng #349: crosstool-ng does not strip nano variant libs

stephanosio, 2021-06-07:
Zephyr crosstool-ng does not strip the nano variant libs (`libc_nano.a`, `libstdc++_nano.a`, ...).

The commit that added this feature is: https://github.com/zephyrproject-rtos/crosstool-ng/commit/5dc17d3bb1781269b7594cd643a32b316ffcef3c

It is likely that this feature initially worked, but later stopped working after updating to the latest crosstool-ng with different nano variant lib support patches merged upstream.

stephanosio, 2021-06-07:
cc @galak

stephanosio, 2021-07-13:
It looks like I've already fixed this earlier:
https://github.com/zephyrproject-rtos/crosstool-ng-old/commit/8c79efc88232af6d51b3a9d9c95058e0fd606ff4

galak, 2021-07-14:
I pulled that commit into crosstool-ng SDK-0.13.0-test

galak, 2021-07-14:
PR #356

galak, 2021-07-14:
@stephanosio can you send a upstream PR for this to crosstool-ng

stephanosio, 2021-07-14:
@galak will do, once we verify that it works in that PR.

stephanosio, 2021-07-18:
With https://github.com/zephyrproject-rtos/crosstool-ng-old/commit/8c79efc88232af6d51b3a9d9c95058e0fd606ff4 (built by #356), the library size looks more sane now:
```
-rw-r--r-- 2 stephanos stephanos 1.3M Jul 15 01:21 aarch64-zephyr-elf/aarch64-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 1.4M Jul 15 01:21 aarch64-zephyr-elf/aarch64-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 4.6M Jul 15 01:21 aarch64-zephyr-elf/aarch64-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 4.9M Jul 15 01:21 aarch64-zephyr-elf/aarch64-zephyr-elf/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 1.3M Jul 15 02:12 arc64-zephyr-elf/arc64-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 1.4M Jul 15 02:12 arc64-zephyr-elf/arc64-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 4.3M Jul 15 02:12 arc64-zephyr-elf/arc64-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 5.0M Jul 15 02:12 arc64-zephyr-elf/arc64-zephyr-elf/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 1.1M Jul 15 01:40 arc-zephyr-elf/arc-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 1.1M Jul 15 01:40 arc-zephyr-elf/arc-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 3.3M Jul 15 01:40 arc-zephyr-elf/arc-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 3.7M Jul 15 01:40 arc-zephyr-elf/arc-zephyr-elf/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 1.1M Jul 15 01:55 arm-zephyr-eabi/arm-zephyr-eabi/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 1.1M Jul 15 01:55 arm-zephyr-eabi/arm-zephyr-eabi/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 3.5M Jul 15 01:55 arm-zephyr-eabi/arm-zephyr-eabi/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 5.4M Jul 15 01:55 arm-zephyr-eabi/arm-zephyr-eabi/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 1.4M Jul 15 02:17 mips-zephyr-elf/mips-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 1.4M Jul 15 02:17 mips-zephyr-elf/mips-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 3.8M Jul 15 02:17 mips-zephyr-elf/mips-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 4.2M Jul 15 02:17 mips-zephyr-elf/mips-zephyr-elf/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 992K Jul 15 01:22 nios2-zephyr-elf/nios2-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 994K Jul 15 01:22 nios2-zephyr-elf/nios2-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 3.5M Jul 15 01:22 nios2-zephyr-elf/nios2-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 3.8M Jul 15 01:22 nios2-zephyr-elf/nios2-zephyr-elf/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 1.5M Jul 15 01:23 riscv64-zephyr-elf/riscv64-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 1.5M Jul 15 01:23 riscv64-zephyr-elf/riscv64-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 4.6M Jul 15 01:23 riscv64-zephyr-elf/riscv64-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 5.5M Jul 15 01:23 riscv64-zephyr-elf/riscv64-zephyr-elf/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 994K Jul 15 01:34 sparc-zephyr-elf/sparc-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 980K Jul 15 01:34 sparc-zephyr-elf/sparc-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 3.4M Jul 15 01:34 sparc-zephyr-elf/sparc-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 3.7M Jul 15 01:34 sparc-zephyr-elf/sparc-zephyr-elf/lib/libstdc++.a
-rw-r--r-- 2 stephanos stephanos 1.6M Jul 15 01:23 x86_64-zephyr-elf/x86_64-zephyr-elf/lib/libc.a
-rw-r--r-- 1 stephanos stephanos 1.3M Jul 15 01:23 x86_64-zephyr-elf/x86_64-zephyr-elf/lib/libc_nano.a
-rw-r--r-- 1 stephanos stephanos 4.2M Jul 15 01:23 x86_64-zephyr-elf/x86_64-zephyr-elf/lib/libstdc++_nano.a
-rw-r--r-- 1 stephanos stephanos 4.6M Jul 15 01:23 x86_64-zephyr-elf/x86_64-zephyr-elf/lib/libstdc++.a
```

This patch is based on the earlier patch (not upstreamed) https://github.com/zephyrproject-rtos/crosstool-ng/commit/5dc17d3bb1781269b7594cd643a32b316ffcef3c, and we should re-evaluate its approach before upstreaming it as per the previous review comments: see https://github.com/crosstool-ng/crosstool-ng/pull/1279#issuecomment-562816540.

TODO:
1. Evaluate using `strip --strip-debug` instead of manually specifying the sections to be removed -- on second thought, `.debug_frame` should not really matter since newlib does not make use of any "exception handling."
2. Consider merging `CT_STRIP_TARGET_TOOLCHAIN_LIBRARIES` and `CT_STRIP_TARGET_TOOLCHAIN_EXECUTABLES` since, although not using the same mechanism, they achieve the same thing.

closed 2021-07-19 by galak