zephyrproject-rtos/sdk-ng #350: Add Zephyr-specific target for toolchain components

stephanosio, 2021-07-01:
The Zephyr SDK toolchain components are currently built targeting the generic bare metal target (i.e. `arm-*-eabi` for AArch32 and `(arch)-*-elf` for the rest).

The generic bare metal target, as its name suggests, lacks proper OS-specific customisation and is therefore very limited in its capability to support the features that require close integration with the target operating system (e.g. synchronisation primitives, reentrancy, support for the standard library functions that require OS-specific implementation).

**Here I propose to add the Zephyr RTOS-specific targets named `(arch)-zephyr` for the purpose of supporting more advanced features that require close OS integration.**

The SDK toolchain components of interest at this time are the following:
* GNU Binutils
* GNU Compiler Collection (GCC)
* Newlib
* GNU C++ Library (part of GCC; aka. libstdc++)

The primary goal of this is as follows:
* Support reentrancy in binutils
* Support thread safety and reentrancy in libstdc++
* Support OS-specific primitives in libstdc++

stephanosio, 2021-07-01:
cc @galak @nashif @tejlmand @carlescufi

stephanosio, 2021-12-10:
**Re: libstdc++ thread safety**

* libstdc++ internally uses the gthread implementation provided by the libgcc.
* Support `--enable-threads=zephyr`

gthread interface definition: https://github.com/gcc-mirror/gcc/blob/16e2427f50c208dfe07d07f18009969502c25dc8/libgcc/gthr.h
gthread interface implementation for RTEMS: https://github.com/gcc-mirror/gcc/blob/16e2427f50c208dfe07d07f18009969502c25dc8/libgcc/config/gthr-rtems.h
std::mutex implementation using gthread: https://github.com/gcc-mirror/gcc/blob/16e2427f50c208dfe07d07f18009969502c25dc8/libstdc%2B%2B-v3/include/bits/std_mutex.h
mutex: https://github.com/gcc-mirror/gcc/blob/16e2427f50c208dfe07d07f18009969502c25dc8/libstdc%2B%2B-v3/include/std/mutex

cfriedt, 2022-02-23:
> **Re: libstdc++ thread safety**
>
> * libstdc++ internally uses the gthread implementation provided by the libgcc.
> * Support `--enable-threads=zephyr`
>
> gthread interface definition: [...]

I started trying to get C++ thread and mutex support working some time ago. One of the major challenges (especially for std::thread) is the idea of dynamically created (user) thread stacks.

I'm 100% interested in helping to support this feature.

chrisc11, 2022-02-25:
For newlib, just wanted to note that one other item that would be very useful if a version winds up getting compiled specifically for Zephyr would be emit debug information by adding the -g cflag to the compilation. Many times in the past, I've seen the lack of this flag (and re-entrancy) be big pain points for other targets like the newlib bundled with GNU Arm Embedded Toolchain releases.

This would not impact the final binary size since all debug info gets stripped for .bin generation and would greatly improve debug-ability by improving stack unwinds for issues that take place while in a newlib callchain and being able to easily decode internal newlib types.

lgl88911, 2022-04-03:
@stephanosio May I ask if there is a specific completion time for "Add Zephyr RTOS targets to Newlib", and once it is completed does it mean that we can directly use newlib's posix instead of zephyr's posix subsystem? Thanks!

stephanosio, 2022-04-05:
> May I ask if there is a specific completion time for "Add Zephyr RTOS targets to Newlib"

No timeline yet; but, this will likely be implemented before this year ends.

> once it is completed does it mean that we can directly use newlib's posix instead of zephyr's posix subsystem?

More like in conjunction with the Zephyr's POSIX subsystem. The newlib Zephyr port will be designed such that there are no conflicts with the Zephyr's POSIX subsystem, and also that it makes use of the Zephyr subsystem(s) where possible.

lgl88911, 2022-04-09:
@stephanosio  Thanks for reply!

> More like in conjunction with the Zephyr's POSIX subsystem. The newlib Zephyr port will be designed such that there are no conflicts with the Zephyr's POSIX subsystem, and also that it makes use of the Zephyr subsystem(s) where possible.
Got it. When I want to link zephyr as a library and other source code files into an executable, even if the other files only use the posix API, I can't get rid of the dependency on the zephyr kernel header files.

cfriedt, 2022-04-11:
@stephanosio - I would be happy to get the minimal libc in better shape to support POSIX items, although I believe Newlib is better for that. It still helps to have a relatively small c library in-tree.

I do agree that the minimal libc should be mostly header-compatible with newlib (see PR below, for example). However, I do not believe that Newlib needs to change much, if at all, to support POSIX in Zephyr.

Also `std::thread` and `std::mutex` are on my agenda. Just very slow-moving (see PR below).

Relevant PRs:

- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/44727
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/43987
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/43729

kedareswararao, 2026-09-18:
We've been working this same gap for C++ on AMD Cortex targets (R5/R52/A53/A72/A78)
  and have hardware-validated data that maps directly onto the plan in this issue,
  so wanted to share it and offer to help move it forward.
  **Confirming the problem end-to-end.** The stock SDK `arm-zephyr-eabi` /
  `aarch64-zephyr-elf` toolchains are `Thread model: single`, so libstdc++ selects
  `std::__default_lock_policy == _S_single` and `std::shared_ptr` refcounting compiles
  to a non-atomic `ldr; sub; str` (no `LDREX/STREX`). Under `CONFIG_MULTITHREADING`
  this corrupts the control block and crashes in `_Sp_counted_base::_M_release()`,
  even on a single core; C++ exception state is likewise not thread-isolated. Verified
  identical across all five AMD cores (the R5/R52 share the same 32-bit libstdc++, and
  A53/A72/A78 share `aarch64-zephyr-elf`). This is the same root cause tracked in
  zephyr#105427.
  **Following @cfriedt's proposed path (gthr-posix first, gthr-zephyr later).** We took
  exactly that first step out-of-tree and validated it on real silicon:
  - Rebuilt both AMD toolchains with `--enable-threads=posix --enable-tls
    --enable-libstdcxx-threads` (crosstool-NG over the sdk-ng GCC 14.3.0 sources),
    binding libgcc's `gthr-posix.h` to Zephyr's POSIX layer.
  - Static result flips as expected: `Thread model: posix`, `__GTHREADS` defined,
    `__default_lock_policy == _S_atomic`, and `_M_release` now emits real atomics
    (`ldrex/strex` on R5, `ldaddal` on A78).
  - Runtime, on hardware: `std::shared_ptr` refcount stress test PASSes with the
    toolchain change **alone**; cross-thread C++ exception isolation PASSes with the
    toolchain plus the in-flight Zephyr pthread PRs (#109291 / #112736 / #112960 /
    #113166). Validated on Cortex-R5 (VCK190) and on Cortex-A78 4-core SMP (VersalNet).
  So the gthr-posix milestone in this issue is demonstrably working, and the split is
  clean: the `shared_ptr` fix is pure-toolchain; the exception-handling case needs the
  Zephyr-side pthread work on top.
  **How we'd like to help.** We're not looking to open a competing effort — we'd like
  to help land the in-tree version of what's already outlined here, and are happy to
  do whatever it takes to enable this support. A few questions to avoid duplicating
  work:
  1. For the near-term gthr-posix milestone: what's the best home for the
     crosstool-NG config change in `sdk-ng` (flipping the `arm-zephyr-eabi` /
     `aarch64-zephyr-elf` targets to `threads=posix` + the pthread header wiring)?
  2. For the eventual `gthr-zephyr` optimization: is the Centrinix binutils work still
     the intended base, and is there any GCC/libgcc `gthr-zephyr.h` starting point, or
     is that still greenfield?
  3. Would standing AMD hardware CI (R5 + A78 SMP) running the reproducer be useful as
     validation for whatever lands?
  Let us know your thoughts on where we can plug in.

cfriedt, 2026-09-18:
@kedareswararao - are to able to update the pr links in your comment so that they appear as links? You can copy the entire URL for each link into your comment, or use github's shortened notation for cross-referencing PRs.

kedareswararao, 2026-09-18:
> [@kedareswararao](https://github.com/kedareswararao) - are to able to update the pr links in your comment so that they appear as links? You can copy the entire URL for each link into your comment, or use github's shortened notation for cross-referencing PRs.

Sorry @cfriedt updated now