zephyrproject-rtos/sdk-ng #354: Use Zephyr forks of binutils/gcc/gdb

stephanosio, 2021-07-13:
#### Summary

At the time of writing, the binutils, gcc and gdb components are built from the released source archive with the sdk-ng-side patches applied on top of them.

The current approach makes it difficult to manage and keep track of the patches, so consider using Zephyr fork like [we currently do with the newlib](https://github.com/zephyrproject-rtos/newlib-cygwin).

#### Implementation Details

For binutils-gdb, use the [Zephyr binutils-gdb repository](https://github.com/zephyrproject-rtos/binutils-gdb).
For gcc, use the [Zephyr gcc repository](https://github.com/zephyrproject-rtos/gcc).

1. For every upstream release to be forked, create a branch (e.g. `zephyr-gcc-10` for GCC 10 release).
2. Commit Zephyr-specific patches in the fork branch.
3. Use the fork branch in sdk-ng to build the relevant component.

stephanosio, 2021-07-13:
cc @galak

galak, 2021-07-13:
@stephanosio in general no issue with this.  How do we deal with ARC64 having some unique patches?

stephanosio, 2021-07-13:
> How do we deal with ARC64 having some unique patches?

I suppose those patches are going to be upstreamed eventually -- in that case, that shouldn't be a problem in the long term.

As for now, is there a reason to make those patches arc64-specific (i.e. only applied for the arc64 build)? Are those patches not orthogonal to other archs?

I see that we are already globally applying those patches for [gcc](https://github.com/zephyrproject-rtos/sdk-ng/blob/2686e98931e0f229de01dc4511d399e91288a649/patches-arc64/gcc#L1) and [gdb](https://github.com/zephyrproject-rtos/sdk-ng/blob/2686e98931e0f229de01dc4511d399e91288a649/patches-arc64/gdb#L1), so those won't be a problem. Only [binutils has an arc64-specific patch](https://github.com/zephyrproject-rtos/sdk-ng/tree/main/patches-arc64/binutils/2.35.1).

Although I haven't done an in-depth analysis on [the binutils patch](https://raw.githubusercontent.com/zephyrproject-rtos/sdk-ng/main/patches-arc64/binutils/2.35.1/0001-Add-ARC64-support.patch), it seems to be fairly orthogonal to other archs (except arc32, maybe).

cc @abrodkin

galak, 2021-07-13:
https://github.com/zephyrproject-rtos/sdk-ng/commit/0a4322d161d9ddd796a2304352f1a64e3a4a1735 comments on the ARC64 binutils patch possibly breaking things and thus it being unique.

galak, 2021-07-13:
@stephanosio also, how do we handle the patches in crosstool-ng.

galak, 2021-07-13:
@stephanosio also, FYI for gcc:

https://github.com/zephyrproject-rtos/gcc/tree/zephyr-10.3.0

stephanosio, 2021-07-13:
> [0a4322d](https://github.com/zephyrproject-rtos/sdk-ng/commit/0a4322d161d9ddd796a2304352f1a64e3a4a1735) comments on the ARC64 binutils patch possibly breaking things and thus it being unique.

In that case, we can keep an ARC64-specific patch directory on the sdk-ng side until they are merged upstream.

galak, 2021-07-13:
> In that case, we can keep an ARC64-specific patch directory on the sdk-ng side until they are merged upstream.

I can't remember will crosstool apply a patch to a GIT repo?

stephanosio, 2021-07-13:
> I can't remember will crosstool apply a patch to a GIT repo?

will have to check, but IIRC, for `CT_LOCAL_PATCH_DIR` it does.

For the patches under `packages/` (e.g. https://github.com/crosstool-ng/crosstool-ng/tree/master/packages/gcc/10.3.0), if we use `CT_*_DEVEL_URL` for obvious reasons.

> also, how do we handle the patches in crosstool-ng.

so, for those, we will need to pull in those patches into the Zephyr fork as well.

stephanosio, 2021-07-13:
> will have to check, but IIRC, for CT_LOCAL_PATCH_DIR it does.

So, when using `CT_*_DEVEL_URL`, it IS still possible to provide local patches using `CT_LOCAL_PATCH_DIR` by placing the patches under `${CT_LOCAL_PATCH_DIR}/(packagename)/git-(commit)` directory:

https://github.com/zephyrproject-rtos/crosstool-ng-old/tree/master-stephanos/patches/zephyr/newlib

For instance, `${CT_LOCAL_PATCH_DIR}/gcc/git-b06f1b57/*.patch`.

The only problem here is that the git commit sha is used in the patch directory name, so the directory name will need to be updated whenever there is a new head revision in the fork.

I wonder if there is a better way to handle this ... if not, we could probably implement a crosstool-ng feature that allows using branch name instead of commit sha in the patch directory name.

For example, if we have:
```
CT_LOCAL_PATCH_DIR="${CT_TOP_DIR}/../../patches-arc64"
CT_BINUTILS_DEVEL_URL="https://github.com/zephyrproject-rtos/binutils-gdb.git"
CT_BINUTILS_DEVEL_BRANCH="zephyr-binutils-2.35"
```

We place the patches under `patches-arc64/binutils/git-zephyr-binutils-2.35/`.

stephanosio, 2021-07-13:
The patch issue will be addressed by https://github.com/crosstool-ng/crosstool-ng/pull/1560.

With the patch, we can simply place the patch under `patches-arc64/binutils/` without version-specific directory.

closed 2021-07-14 by galak

abrodkin, 2021-07-15:
@galak @stephanosio can we just add a separate branches for ARC64 in Zephyr's GCC & Binutils-GDB repos and just use corresponding SHA's in CT-NG building for ARC64? That IMHO will simplify maintenance a bit as those patches are (a) quite large & (b) will be changing a lot in between snapshots as it's still quite heavily WIP still.

stephanosio, 2021-07-15:
@abrodkin Do you have an estimate on when those patches will be available upstream? If it is going to take a long time, I agree that having separate ARC64 branches would be better, under the condition that the ARC64 branches will be kept rebased strictly on top of the relevant Zephyr branches.

abrodkin, 2021-07-15:
@stephanosio I don't have any good estimations because we're yet to start upstreaming of all the toolchain components and I guess you know that upstream review might take quite some time and as contributors we don't have any way to push reviewers. So I'd say we may start upstreaming in a couple of months from now and hopefully if all goes smoothly we'll get it accepted say in a year or so. Still we're obviously late for GCC 12.x and new architecture being the major change it might only be introduced in GCC 13.x which will only see light of day in early 2023. So that's regarding a possible schedule.
As for branches used as a base for ARC64 changes I do understand your concern and we'll try to match it but given there're 2 completely separate processes (Zephyr switching versions of upstream components in some points in time & Synopsys re-basing its changes on top of upstream releases) I'm not sure we may guarantee those 2 to be all the time in sync. But we may try :)

PR #355 (2021-07-14): Pull in Zephyr binutils/gcc/gdb forks
Update configs to pull in Zephyr binutils/gcc/gdb forks.

The Zephyr-specific patches that were previously provided as local patches in the sdk-ng were merged into the relevant Zephyr forks below:

binutils: https://github.com/zephyrproject-rtos/binutils-gdb/tree/zephyr-binutils-2_35_1
gcc: https://github.com/zephyrproject-rtos/gcc/tree/zephyr-gcc-10.3.0
gdb: https://github.com/zephyrproject-rtos/binutils-gdb/tree/zephyr-gdb-9.2

As for the crosstool-ng-provided patches (i.e. bundled/package patches), they have been committed as single "crosstool-ng patch" [... cut]