zephyrproject-rtos/sdk-ng #369: Footprint increase

stephanosio, 2021-07-23:
A significant footprint increase has been reported for the SDK 0.13.0-* releases.

For `samples/net/sockets/coap_client` with `nrf52840dk_nrf52840`

**0.12.4**
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      116016 B         1 MB     11.06%
            SRAM:       28320 B       256 KB     10.80%
        IDT_LIST:          0 GB         2 KB      0.00%
```

**0.13.0-rc3**
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      143080 B         1 MB     13.65%
            SRAM:       30336 B       256 KB     11.57%
        IDT_LIST:          0 GB         2 KB      0.00%
```

This increase happened after the commits in the PR #333, which switched to using the upstream crosstool-ng newlib nano build process.

stephanosio, 2021-07-23:
cc @galak

stephanosio, 2021-07-28:
**Comparing newlib nano build parameters for `arm-zephyr-eabi`**

[Good](https://buildkite.com/zephyr/sdk-ng/builds/258#42167e87-dcf1-4de7-9dbb-ccd26ca17345)
```
[EXTRA]    Configuring C library (nano)
[DEBUG]    ==> Executing:
CC_FOR_BUILD='x86_64-build_pc-linux-gnu-gcc'
CFLAGS_FOR_TARGET=' -mthumb-interwork   -ffunction-sections -fdata-sections'
AR_FOR_TARGET='/workdir/build/build_arm/.build/arm-zephyr-eabi/buildtools/bin/arm-zephyr-eabi-gcc-ar'
RANLIB_FOR_TARGET='/workdir/build/build_arm/.build/arm-zephyr-eabi/buildtools/bin/arm-zephyr-eabi-gcc-ranlib'
'/bin/bash'
'/workdir/build/build_arm/.build/arm-zephyr-eabi/src/newlib/configure'
'--host=x86_64-build_pc-linux-gnu'
'--target=arm-zephyr-eabi'
'--prefix=/workdir/build/build_arm/.build/arm-zephyr-eabi/build/build-libc-nano/target-libs'
'--enable-newlib-io-float'
'--disable-newlib-io-long-double'
'--disable-newlib-supplied-syscalls'
'--disable-newlib-io-pos-args'
'--disable-newlib-io-c99-formats'
'--disable-newlib-io-long-long'
'--disable-newlib-register-fini'
'--enable-newlib-nano-malloc'
'--enable-newlib-nano-formatted-io'
'--enable-newlib-atexit-dynamic-alloc'
'--enable-newlib-global-atexit'
'--enable-lite-exit'
'--enable-newlib-reent-small'
'--disable-newlib-multithread'
'--enable-newlib-retargetable-locking'
'--disable-newlib-wide-orient'
'--disable-newlib-fseek-optimization'
'--enable-newlib-fvwrite-in-streamio'
'--disable-newlib-unbuf-stream-opt'
'--enable-target-optspace'
```

[Bad](https://buildkite.com/zephyr/sdk-ng/builds/282#6f2b11da-a0a0-4595-a4de-7ce40fdf2dfd)
```
[EXTRA]    Configuring Newlib Nano library
[DEBUG]    ==> Executing:
CC_FOR_BUILD='x86_64-build_pc-linux-gnu-gcc'
CFLAGS_FOR_TARGET='   -ffunction-sections -fdata-sections'
AR_FOR_TARGET='/workdir/build/output/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc-ar' 
RANLIB_FOR_TARGET='/workdir/build/output/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc-ranlib'
'/bin/bash' 
'/workdir/build/build_arm/.build/arm-zephyr-eabi/src/newlib-nano/configure'
'--host=x86_64-build_pc-linux-gnu'
'--target=arm-zephyr-eabi' '--prefix=/workdir/build/output/arm-zephyr-eabi'
'--exec-prefix=/workdir/build/output/arm-zephyr-eabi/newlib-nano'
'--libdir=/workdir/build/output/arm-zephyr-eabi/newlib-nano/arm-zephyr-eabi/lib'
'--enable-newlib-io-float'
'--disable-newlib-io-long-double'
'--disable-newlib-supplied-syscalls'
'--disable-newlib-io-pos-args'
'--disable-newlib-io-c99-formats'
'--disable-newlib-io-long-long'
'--disable-newlib-register-fini'
'--enable-newlib-nano-malloc'
'--enable-newlib-nano-formatted-io'
'--enable-newlib-atexit-dynamic-alloc'
'--enable-newlib-global-atexit'
'--enable-lite-exit'
'--enable-newlib-reent-small'
'--disable-newlib-multithread'
'--enable-newlib-retargetable-locking'
'--disable-newlib-wide-orient'
'--disable-newlib-fseek-optimization'
'--enable-newlib-fvwrite-in-streamio'
'--disable-newlib-unbuf-stream-opt'
'--enable-target-optspace'
```

galak, 2021-07-28:
Looks like maybe `-mthumb-interwork `?

galak, 2021-07-28:
https://github.com/crosstool-ng/crosstool-ng/issues/1566

stephanosio, 2021-07-28:
As I thought, it's the [new `nano.specs`](https://github.com/crosstool-ng/crosstool-ng/blob/21c68181268214a1790f61aadd6ab3d8c55f2002/scripts/build/companion_libs/350-newlib_nano.sh#L199-L213) that is causing this.

I copied `arm-zephyr-eabi/lib/nano.specs` from "good" to "bad," and the footprint came back to normal:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      116016 B         1 MB     11.06%
            SRAM:       28320 B       256 KB     10.80%
        IDT_LIST:          0 GB         2 KB      0.00%
```

Diff between "good' and "bad" `nano.specs`:
```diff
@@ -1,13 +1,22 @@
-%rename link   newlib_nano_link
-%rename cpp    newlib_nano_cpp
-%rename cc1plus        newlib_nano_cc1plus
+%rename link                nano_link
+%rename link_gcc_c_sequence                nano_link_gcc_c_sequence
+%rename cpp_unique_options             nano_cpp_unique_options

-*cpp:
--isystem %:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-zephyr-eabi/include) %(newlib_nano_cpp)
+*cpp_unique_options:
+-isystem =/include/newlib-nano %(nano_cpp_unique_options)

-*cc1plus:
--idirafter %:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-zephyr-eabi/include) %(newlib_nano_cc1plus)
+*nano_libc:
+-lc_nano
+
+*nano_libgloss:
+%{specs=rdimon.specs:-lrdimon_nano} %{specs=nosys.specs:-lnosys}
+
+*link_gcc_c_sequence:
+%(nano_link_gcc_c_sequence) --start-group %G %(nano_libc) %(nano_libgloss) --end-group

 *link:
--L%:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-zephyr-eabi/lib/%M) -L%:getenv(GCC_EXEC_PREFIX ../../newlib-nano/arm-zephyr-eabi/lib)
+%(nano_link) %:replace-outfile(-lc -lc_nano) %:replace-outfile(-lg -lg_nano) %:replace-outfile(-lrdimon -lrdimon_nano) %:replace-outfile(-lstdc++ -lstdc++_nano) %:replace-outfile(-lsupc++ -lsupc++_nano)
+
+*lib:
+%{!shared:%{g*:-lg_nano} %{!p:%{!pg:-lc_nano}}%{p:-lc_p}%{pg:-lc_p}}
```

I don't have time to look into it today, will probably further investigate this tomorrow.

galak, 2021-07-28:
@stephanosio good work on tracking that down. I'll take a look to see if I can isolate it further.

stephanosio, 2021-07-29:
https://github.com/crosstool-ng/crosstool-ng/pull/1568

PR #378 (2021-07-30): crosstool-ng: Update to pull in fix for nano.spec
pull in nano.spec fix that links with -lc_nano instead of -lc (same
for other nano lib variants, libm, libstdc++, etc..)

Fixes #369
closed 2021-07-30 by galak