zephyrproject-rtos/sdk-ng #385: Assess feasibility of using Azure Pipelines for macOS build

stephanosio, 2021-08-04:
The toolchain build for the macOS hosts can only be done from macOS, and this requires us to provide the macOS CI agents.

For now, we are using the GitHub Actions for this; but, it is too unstable for lengthy workflows to the degree that it is simply unusable (macOS jobs often get randomly cancelled and you have to manually restart the entire workflow until all jobs finish successfully, and that can take many retries).

Azure Pipelines seems to be a good alternative to the current unstable GitHub Actions because:
1. It supports macOS agents (in fact, it uses the same OS image as the GitHub Actions).
2. It provides up to 10 parallel jobs with unlimited minutes for free (for open source projects).
3. It is the most cost effective option ($40/additional parallel job/month, much cheaper than getting our own cloud macs).

For this reason, assess the feasibility of moving the macOS build process to the Azure Pipelines -- in particular, we will need to check if the macOS agents for the Azure Pipelines are stable enough to be used for the SDK build workflows.

stephanosio, 2021-08-13:
I created [a repository to test macOS builds using Azure Pipelines](https://github.com/stephanosio/zephyr-sdk-ng-azurepipelines-test) -- so far, it has never had any random unexplainable job cancellations.

In general, Azure Pipelines seems to be very stable with Microsoft-hosted macOS agents and can be a replacement for the current unstable GitHub Actions-based macOS CI.

Pipeline runs: https://dev.azure.com/zephyrproject-rtos/My%20sdk-ng%20Test/_build?definitionId=1&_a=summary

stephanosio, 2021-08-13:
@galak @tejlmand @carlescufi any objections/feedback on migrating the current GitHub Actions-based macOS CI workflows to Azure Pipelines for sdk-ng?

stephanosio, 2021-09-10:
@nashif FYI, if we decide to go with the Azure Pipelines for the macOS SDK builds, we will likely need at least 17 parallel jobs (basically what we run right now), and that is 7 extra parallel jobs on top of the free 10 parallel jobs, which would cost us $280/month.

As noted above, GitHub Actions is not an option for this because it is very unstable with the Microsoft-hosted macOS agents, and providing self-hosted macOS agents ourselves would be much more expensive than $40/agent/month. Moreover, [GitHub Actions only supports 5 parallel Microsoft-hosted macOS agents](https://docs.github.com/en/actions/reference/usage-limits-billing-and-administration).

carlescufi, 2021-11-03:
> @galak @tejlmand @carlescufi any objections/feedback on migrating the current GitHub Actions-based macOS CI workflows to Azure Pipelines for sdk-ng?

None at all. You should however raise this in the TSC to ensure proper budgeting for next year.

stephanosio, 2021-11-04:
> You should however raise this in the TSC to ensure proper budgeting for next year.

@nashif Have you had a chance to discuss this in the board meeting?

carlescufi, 2021-11-15:
@stephanosio shouldn't we try and use GitHub Actions now? Since @nashif is working on moving towards that for standard CI, perhaps it's better to standardize on those everywhere?

stephanosio, 2021-11-15:
> @stephanosio shouldn't we try and use GitHub Actions now? Since @nashif is working on moving towards that for standard CI, perhaps it's better to standardize on those everywhere?

GitHub Actions (at least, with Microsoft-hosted runners) is too unstable for macOS SDK build workflow as noted above.

We briefly discussed this in the Toolchain WG meeting today; Anas suggested we use the AWS-hosted macOS runners with GitHub Actions (simply as orchestrator) there.

We will need to see how that works out -- i.e. that GitHub Actions with our own (AWS-hosted) macOS runners do not experience random CI failures and job cancellations that are unacceptable for very lengthy workflows as we have in the SDK build process.

galak, 2021-11-15:
The problem with AWS hosted MacOS github runners is cost.  Not sure if this has gotten any better.  But I think its like $24 anytime you use a single MacOS machine for anytime under a day.

stephanosio, 2021-11-29:
**UPDATE**
As per the discussion in the Toolchain WG meeting today, @nashif will check if it is possible to integrate the AWS-hosted Mac runners into the current GitHub Actions infrastructure. Once that is in place, we can use the GitHub Actions for building and testing the macOS Zephyr SDK.

stephanosio, 2021-12-16:
Closing this issue since this is no longer being considered as a viable option; instead, we will look into using GitHub Actions with self-hosted runners (possibly based on MacStadium Orka).

closed 2021-12-16 by stephanosio