zephyrproject-rtos/sdk-ng #386: zephyr-sdk-0.13.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc don't support 'typeof'!!

novumdun, 2021-08-11:
When I write some code like this:
    _double t7 = 9;
    typeof(t7) t888;_
arm-zephyr-eabi-gcc says:
/home/dengbo/workspaces/zephyr-samples/hello_world/fast_print/fast_print_ext.c:30:5: warning: implicit declaration of function 'typeof' [-Wimplicit-function-declaration]
   30 |     typeof(t7) t888;
      |     ^~~~~~
/home/dengbo/workspaces/zephyr-samples/hello_world/fast_print/fast_print_ext.c:30:15: error: expected ';' before 't888'
   30 |     typeof(t7) t888;
      |               ^~~~~
It looks like arm-zephyr-eabi-gcc don't support "typeof" feature?

stephanosio, 2021-08-11:
`typeof` is not a valid operator in the C standard.

closed 2021-08-11 by stephanosio

novumdun, 2021-08-11:
Ok, thanks.