zephyrproject-rtos/sdk-ng #387: macos: sdk 0.13.0: all gdb binaries report unknown symbol format

cfriedt, 2021-08-11:
Tried debugging a riscv64 binary and it wasn't working. Then, noticed it was the case for each architecture. The same thing happens for arm and x86_64

```
west build -b qemu_riscv64 -t debugserver samples/hello_world
```

```
riscv64-zephyr-elf-gdb ~/workspace/zephyr.elf
GNU gdb (crosstool-NG 1.24.0.378_e011758) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_apple-darwin19.6.0 --target=riscv64-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
/Users/cfriedt/.gdbinit:1: Error in sourced command file:
No symbol table is loaded.  Use the "file" command.
Reading symbols from /Users/cfriedt/workspace/zephyr.elf...
I'm sorry, Dave, I can't do that.  Symbol format `elf64-littleriscv' unknown.
(gdb) quit
```

```
md5sum zephyr-sdk-0.13.0-macos-x86_64-setup.run 
74cfb81215494b173723108673a276b5  zephyr-sdk-0.13.0-macos-x86_64-setup.run
```

stephanosio, 2021-08-11:
Seems to be related to:
https://sourceware.org/bugzilla/show_bug.cgi?id=26949

galak, 2021-08-11:
Looks like we might need to backport:

https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;h=b413232211bf

galak, 2021-08-11:
https://github.com/zephyrproject-rtos/binutils-gdb/tree/backport-macos-fix

needs testing

PR #389 (2021-08-13): gdb: backport MacOS fix for ELF file support
Backport fix from upstream GDB that fixes issue with GDB on MacOS
supporting ELF files correctly.

Fixes #387
commit c47457e2 (2021-08-13): gdb: backport MacOS fix for ELF file support

Backport fix from upstream GDB that fixes issue with GDB on MacOS
supporting ELF files correctly.

Fixes #387
closed 2021-08-13 by galak