zephyrproject-rtos/sdk-ng #393: SDK installer, include version number

koffes, 2021-09-02:
When installing the SDK on Linux, the default install folder is `sdk-zephyr/`.
In the doc, it is recommended to use e.g. `zephyr-sdk-0.13.0/`
https://docs.zephyrproject.org/latest/getting_started/index.html

It would be preferable if the installer set the version of the SDK as the default install path.

To get around this, I have renamed the folder after install. This seems to work on first glance, but did give a rather cryptic dtc error:
```
Could NOT find working dtc: Found dtc
  (/home/XXX/zephyr-sdk-0.13.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
```

followed by `but failed to load with: ` << blank output >>

This folder actually exists and the system seems to default to another dtc which is in path.
If it is not possible to rename after install, there should be some check for this.

stephanosio, 2021-09-02:
I've transferred this issue to `sdk-ng` repository since it is a Zephyr SDK (and not the Zephyr RTOS itself) issue.

koffes, 2021-09-02:
I applaud the efforts in the linked issue, I was not aware of this.
The remainder of this issue then is to have some form of safety so post-install renaming will either work or fail with a descriptive error. If that is being solved/circumvented in the new approach then this ticket may be closed.

closed 2021-09-02 by koffes