zephyrproject-rtos/sdk-ng #395: Distribute SBOM (software bill of materials) with each Zephyr SDK release

stephanosio, 2021-09-06:
_(suggested in the Toolchain WG meeting today by Roberto Bagnara)_

Document and distribute an SBOM (software bill of materials) with each Zephyr SDK release to help users easily identify all included components and their versions.

An SBOM shall:

* only include the top-level components (and not the sub-components) for the purpose of simplification.
    * for instance, SBOM shall include GCC, but not the components used by it such as MPFR and GMP.
* specify the version of each top-level component included in the Zephyr SDK (e.g. binutils 2.35.1, gcc 10.3.0, ...).
* list (or provide a link to the list of) the patches applied on top of the upstream release for each component.

stephanosio, 2021-09-06:
cc @galak

RobertoBagnara, 2021-09-06:
For the purpose of MISRA compliance, all the documentation regarding the implementation-defined behaviors has to be identified.  Typically this includes the manuals of: the C preprocessor (e.g., for GCC, the appropriate version of cpp.pdf), the C compiler (e.g., gcc.pdf), linker (e.g., ld.pdf), assembler, librarian, and the standard library implementation (sometimes the manual of the mathematical library is a separate document).  Ideally, the SDK should contain the documentation.  If this is unfeasible, links to the applicable documentation would be great.  The alternative, error-prone approach of having of each developer or QA person searching for the applicable documentation is suboptimal (and not in line with the spirit of MISRA C Directive 1.1).