zephyrproject-rtos/sdk-ng #399: Mirror/cache sources required to build SDK

stephanosio, 2021-10-02:
As part of the Zephyr SDK build process, crosstool-ng downloads the source tarballs for the required components from external servers (mirror URLs specified under the crosstool-ng `packages/`).

This makes our CI infrastructure depend on the external servers and often leads to CI failures when these servers go down for whatever reason.

Note that we download the source tarballs from many different servers hosted by many different projects/organisations, so the chances of any one of them being down are not so unlikely; in fact, we have seen the failures related to this many times so far.

For instance, in https://buildkite.com/zephyr/sdk-ng/builds/462, CI failed due to the ISL component (required to build gcc) mirror being unavailable:
```
[INFO ]  Retrieving needed toolchain components' tarballs
...
[EXTRA]    Retrieving 'isl-0.24'
[ERROR]    isl: download failed
```

In order to ensure that these failures do not occur, we must mirror the source tarballs for the required components in a server/service managed by or affiliated with the Zephyr Project.

stephanosio, 2021-10-02:
@galak @nashif I think it would be the most hassle-free to just set up an AWS S3 bucket for this. What do you think?

galak, 2021-10-04:
> @galak @nashif I think it would be the most hassle-free to just set up an AWS S3 bucket for this. What do you think?

No issues with the general idea, just need to see how this would work.

stephanosio, 2022-02-10:
TODO: Apply the same strategy for the new CI (#417).

stephanosio, 2022-03-15:
Consider caching source files in the [docker-sdk-build](https://github.com/zephyrproject-rtos/docker-sdk-build) CI image.

PR #453 (2022-03-31): ci: Cache source files
This commit updates the CI workflow to cache the crosstool-ng and poky
source files used for building the Zephyr SDK in the AWS S3, in order
to speed up the builds as well as to improve the availability of the
source files.

Closes #399
closed 2022-03-31 by stephanosio