zephyrproject-rtos/sdk-ng #407: sdk-0.13.1 - arm-zephyr-eabi-gdb depends on libpython3.8

cmorganBE, 2021-10-15:
Ubuntu 21.04 here

```
/home/cmorgan/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb -ex 'target remote :3333' /home/cmorgan/projects/testproject/build/zephyr/zephyr.elf -ex load
/home/cmorgan/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or director
```

There aren't any libpython3.8 packages still available for 21.04.

A few people have tried some work arounds but none of these look particularly good:
- https://blog.lazy-evaluation.net/posts/linux/zephyr-sdk-bullseye.html (could be debian release specific)
- https://launchpad.net/~deadsnakes/+archive/ubuntu/ppa (there are no 3.8 packages for 21.04 here...)

Thoughts?

stephanosio, 2021-10-15:
If you do not need Python scripting capability in GDB, please use `arm-zephyr-eabi-gdb-no-py`.

If you do, you can install third-party Python 3.8 packages (search "ubuntu 21.04 python 3.8" on Google), or install from source.

closed 2021-10-15 by stephanosio

cmorganBE, 2021-10-15:
@stephanosio this worked. I had to figure out to do that with west. Also saw this ticket, https://github.com/zephyrproject-rtos/zephyr/pull/38749 which should make this easier. Maybe someone will support newer pythons at some point too. Anyway, with the below it works:

```
west debug --gdb ~/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb-no-py
```