zephyrproject-rtos/sdk-ng #420: Update Qemu to get MVE support for the MPS3 AN547

theotherjimmy, 2021-12-07:
The mps3 an547, a cortex-m55 cpu with MVE support, fails in the `tests/lib/sprintf/libraries.libc.sprintf` test with incorrect printing. This is caused by incorrectly decoding added-by-MVE shifts as or with PC as a source register. With these shifts decoding as or, the divide by 5 and divide by 10 routines don't work. This has been fixed in upstream qemu, probably related to https://www.mail-archive.com/qemu-devel@nongnu.org/msg827164.html

stephanosio, 2021-12-08:
QEMU does not currently support MVE emulation, and MVE support will be added in the QEMU 6.2 release.

https://wiki.qemu.org/Planning/6.2

QEMU 6.2 release is due in a week and we will update the Zephyr SDK once it is released.

microbuilder, 2022-02-02:
@stephanosio re 6.2.0 ... just a heads up that there is a bug with systick in QEMU 6.2.0 that breaks mps2_an521 and mps3_an547 support, perhaps others. I tested this fix this morning, but we will either need to update to the next release to fix MPSx support, or apply this patch: https://lists.gnu.org/archive/html/qemu-devel/2022-02/msg00333.html

PR #421 (2022-03-21): qemu: Update to QEMU 6.2 release
This commit updates the QEMU version to 6.2.

QEMU 6.2 is required to support ARM M-Profile Vector Extension (MVE)
emulation.

Closes #420.
Closes #423.
closed 2022-03-21 by stephanosio