zephyrproject-rtos/sdk-ng #447: Please pull-in Binutils tree with ARC64 changes based on 2.35.1

abrodkin, 2022-03-21:
@stephanosio here you may find a tag/snapshot of the sources used in SDK-ng 0.13.0 via https://github.com/zephyrproject-rtos/sdk-ng/pull/327.

Also note (as indicated in the aforementioned PR) that particular snapshot is only good for building ARC64 as code was not yet properly merged with ARCv2. I.e. we need to use upstream vanilla 2.35.1 for ARCv2 and that tag for ARC64.

Once you pull-in that tag/branch I'll post a PR for SDK-ng which uses that for ARC64.

But looking forward we'll be able to use the same sources for all ARC flavors and we've goth ARC64 merged with ARCv2 (ARC HS, ARC EM) a while ago, but that code is based on GCC 11.x.

Or maybe we use SNPS GitHub repo similar to QEMU so that we don't need to sync things up every now and then?

stephanosio, 2022-03-21:
We are already pulling in the ARC64 patch with the new scheme (special exception made for ARC64):
https://github.com/zephyrproject-rtos/sdk-ng/blob/563159647114d9e352e2030dfb1686467169d131/configs/arc64-zephyr-elf.config#L15-L17

https://github.com/zephyrproject-rtos/sdk-ng/blob/563159647114d9e352e2030dfb1686467169d131/configs/arc64-zephyr-elf.config#L5

Patch: https://github.com/zephyrproject-rtos/sdk-ng/tree/main/patches-arc64/binutils

> Or maybe we use SNPS GitHub repo similar to QEMU so that we don't need to sync things up every now and then?

QEMU was a special case because we do not/will likely not have global Zephyr-specific patches for it.

As for Binutils and GCC, we plan to implement the `arch-*-zephyr` targets as per #350 and we would not want to patch/rebase onto the Synopsys repo whenever we update the Zephyr targets.

abrodkin, 2022-03-25:
@stephanosio ooops missed your question in the end.

We do have now both GCC & Binutils sources with properly integrated ARCv3 code (i.e. the same source tree meant to be used for both ARCompact/ARCv2 & ARCv3). E.g. Binutils/GDB tree is here https://github.com/foss-for-synopsys-dwc-arc-processors/binutils-gdb/tree/arc64. But it is now based on top of the very recent Binutils/GDB upstream sources (https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=0b35f123c200522998a81335dc218551ca7b3d92).

And I need to check how hard it might be to back-port ARCv3 changes on top of the most recent upstream releases of both Binutils & GDB. @shahab-vahedi @cupertinomiranda could you guys please comment on that feasibility?

stephanosio, 2022-03-25:
> But it is now based on top of the very recent Binutils/GDB upstream sources

If that is the case, how about we wait a few months before doing that?

There is a plan to update Binutils/GCC/GDB  in the 2022 H1 (https://github.com/zephyrproject-rtos/sdk-ng/issues/408).

stephanosio, 2022-03-25:
Based on #409, the backport target versions should be:

* Binutils 2.38 (already released)
* GCC 12.1 (scheduled for April 2022)
* GDB 12.1 (likely to be released in June 2022) or 11.2 (if 12.1 gets delayed)

abrodkin, 2022-03-25:
Well, my main rationale for ARC tools update sometime soon is a need to have HS5x (ARCv3 32-bit processors as opposed to 64-bit HS6x) support which is not there in what we have in the SDK-ng 0.13.x-0.14.1. And that in its turn is a prerequisite for adding HS5x support in upstream Zephyr, which we'd really like to get into v3.1 as it's been functionally ready for quite some time and @evgeniy-paltsev was about to start publishing PR's with that support in Zephyr RTOS.

Sorry for all that mess ;)

stephanosio, 2022-03-25:
@abrodkin Do we need both Binutils and GCC update for HS5x? or Binutils only?

abrodkin, 2022-03-25:
@stephanosio we do need both as back in the day (almost a year ago) HS5x support was not added anywhere, sorry.

stephanosio, 2022-05-01:
Closing in favour of #409

closed 2022-05-01 by stephanosio