zephyrproject-rtos/sdk-ng #452: Please fork and include mcumgr-cli

ddavidebor, 2022-03-30:
Please fork mcumgr inside of the Zephyr project, then include it as a tool in sdk-ng.

This way it will promote contributing to an essential tool of the zephyr ecosystem that has gone almost unmaintained.

stephanosio, 2022-03-31:
@de-nordic @carlescufi @nvlsianpu @lairdjm

As mcumgr maintainers/collaborators, do you think this is something necessary and/or useful?

stephanosio, 2022-03-31:
[As per the discussion in #mcumgr](https://discord.com/channels/720317445772017664/884732002924908584/958982076827320320), there is a plan to implement a new mcumgr utility in Python that replaces the current `mynewt-mcumgr-cli`, so we are going to implement this.

closed 2022-03-31 by stephanosio

de-nordic, 2022-03-31:
We lack force to go through Features/Enhancements/RFCs that are already reported here https://github.com/zephyrproject-rtos/zephyr/projects/37, we did not have go surgeon, on Zephyr side, to take care of the mcumgr-cli when mcumgr was not a part of Zephyr, and that still was essential tool.
I do not think that pulling it in would improve anything - people were adding custom commands to SMP for their apps and did not update the mcumgr-cli to reflect that anyway.