zephyrproject-rtos/sdk-ng #462: add SOF rimage to SDK

nashif, 2022-04-15:
Provide https://github.com/thesofproject/rimage to be part of the SDK to enable building and signing images expected by Intel ADSP targets. This will make the process much easier and would enable creating ready to "flash" images from zephyr, without having to build and call external tools.

stephanosio, 2022-04-19:
Can `rimage` be built for all Linux (AArch64, x86-64), macOS (AArch64, x86-64) and Windows (x86-64) hosts? Has this been tested?

nashif, 2022-04-19:
afaik yes, it is used on windows for sure.

stephanosio, 2022-04-19:
> We have not included other such tools for other platforms in the past.

well, although its purpose is somewhat different, we do currently have `BOSSA` which is used to flash only the Atmel SoCs in the SDK.

Xilinx QEMU and the recently added ARC QEMU are another examples of vendor-specific utilities that are included in the SDK.

stephanosio, 2023-05-31:
This should be closed if https://github.com/zephyrproject-rtos/zephyr/issues/58212 is implemented.

closed 2025-02-22 by nashif