zephyrproject-rtos/sdk-ng #463: What is necessary to include and integrate a toolchain in Zephyr SDK?

amarkovicsa, 2022-04-22:
Hello all. I have a question.

I was glad that support for mips toolchain was recently introduced in Zephyr SDK.

However, I am interested more in nanoMips ISA support. It is a variant of Mips ISA that is specifically designed for resource-constrained systems.It is not supported in regular mips toolchains. Iy has a separately available toolchain:

https://github.com/MediaTek-Labs/nanomips-gnu-toolchain/releases

Is it feasible to include that nanoMips toolchain in Zephyr SDK? (in a way similar to various xtensa toolchains)

What would be necessary conditions in general for a toolchain to be included in Zephyr SDK?

Thanks in advance for consideration!

stephanosio, 2022-04-24:
If the nanoMIPS support has been already upstreamed to the relevant projects, we should be able to simply enable the nanoMIPS target(s) when building the toolchains.

If not, you will need to provide the nanoMIPS-specific patches that can be appplied to the current version of Zephyr toolchain components:

* https://github.com/zephyrproject-rtos/binutils-gdb
* https://github.com/zephyrproject-rtos/gcc

Another option is to not include the nanoMIPS support in the Zephyr SDK, and instead add the support for the nanoMIPS GNU toolchain support in the Zephyr build system.

amarkovicsa, 2022-04-25:
I truly appreciate your clarification, @stephanosio. It appears to me that, at least for the time being, the third (last) option is the simplest and the most feasible. I will consult nanoMips toolchain authors as well. Thanks again!

stephanosio, 2022-05-01:
Closing since the question has been answered.

closed 2022-05-01 by stephanosio