zephyrproject-rtos/sdk-ng #483: RV32E stack alignment

carlocaione, 2022-05-22:
@npitre @nordic-krch @carlescufi @stephanosio @pabigot 

What follows could be pure speculation about the issues I'm seeing but I want to raise this nonetheless. You folks are definitely more knowledgeable than me on this topic.

I'm adding support for RV32E. From the compiler prospective everything is the same as for RV32I but I'm now using `-mabi=ilp32e -march=rv32emac`.

I have everything ready and only two tests failing: `tests/subsys/logging/log_msg2/` and `tests/lib/cbprintf_package`.

They fail almost for the same reason:
```
START - test_log_msg2_various_args                                                                        
Test string:%d %d %lld %p %lld %p
-----------------------printing message--------------------
message 0x8000cb98      
package len: 44, data len: 0
0b 00 00 00 bc 96 00 80 fb ff ff ff 45 00 00 00 12 34 56 34 12 34 12 00 e0 cc 00 80 00 00 00 00 12 34 56 34 12 34 12 00 78 99 00 80 
source: 0x7b                                                                                              
timestamp: 287454020                                                                                      
-------------------end of printing message-----------------
-----------------------printing message--------------------
message 0x8000ca98      
package len: 44, data len: 0
0b 00 00 00 bc 96 00 80 fb ff ff ff 45 00 00 00 12 34 56 34 12 34 12 00 e0 cc 00 80 00 00 00 00 12 34 56 34 12 34 12 00 78 99 00 80 
source: 0x7b                                                                                              
timestamp: 287454020                                                                                      
-------------------end of printing message-----------------
-----------------------printing message--------------------
message 0x8000c998      
package len: 44, data len: 0
0b 00 00 00 bc 96 00 80 fb ff ff ff 45 00 00 00 12 34 56 34 12 34 12 00 e0 cc 00 80 00 00 00 00 12 34 56 34 12 34 12 00 78 99 00 80 
source: 0x7b                                                                                              
timestamp: 287454020                                
-------------------end of printing message-----------------        
                                                     
    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_msg2/src/main.c:105: basic_validate: (rv not equal to 0)
expected:                                                                                                 
-5 69 5123802372912146 0x8000cce0 5123802372912146 0x80009978,
got:                                              
-5 69 5123802372912146 0x8000cce0 3771258989883424768 0x123412     
 FAIL - test_log_msg2_various_args in 0.19 seconds
```
and
```
START - test_cbprintf_package
-----------------------------------------
test long b1b2b3b4 bdc5ca39 e4e3e2e1
runtime package 0x8000d130:
05 00 00 00 48 95 00 80 b4 b3 b2 b1 39 ca c5 bd e1 e2 e3 e4 
static package 0x8000d190:
05 00 00 00 48 95 00 80 b4 b3 b2 b1 39 ca c5 bd e1 e2 e3 e4 
-----------------------------------------
test long long b1b2b3b4 1122334455667788 e4e3e2e1
runtime package 0x8000d128:
07 00 00 00 64 96 00 80 b4 b3 b2 b1 00 00 00 00 88 77 66 55 44 33 22 11 e1 e2 e3 e4 

    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cbprintf_package/src/main.c:55: unpack: (strcmp(buf->buf, compare_buf) not equal to 0)
Strings differ
exp: |test long long b1b2b3b4 1122334455667788 e4e3e2e1|
got: |test long long b1b2b3b4 5566778800000000 11223344|

 FAIL - test_cbprintf_package in 0.12 seconds

```

In both cases something is wrong with the packaging of `long long int`.

The only clue I have so far is that while RV32 requires a 16 bytes alignment for the stack pointer, this is not respected when compiling for RV32E. 

For example, for RV32E:
```
int cbvprintf(cbprintf_cb out, void *ctx, const char *fp, va_list ap) 
{
800025e6:       f6810113                addi    sp,sp,-152 
...
```
for RV32I:
```
int cbvprintf(cbprintf_cb out, void *ctx, const char *fp, va_list ap)
{ 
800024bc:       7171                    addi    sp,sp,-176 
...
```
I'm not exactly sure if this is the root cause of the problem I'm seeing with the two tests, but `cbprintf_via_va_list` and `cbprintf_internal.h` seems to rely on some kind of ABI that could be different when compiling for RV32I and RV32E (and it kinda reminds me of https://github.com/riscv-collab/riscv-gcc/issues/63).

So, assuming that this is indeed a problem and the cause of the issues I'm seeing, I think we have two ways forward:
- We fix (?) the SDK so that we can have a proper stack alignment also for RV32E
- We fix (?) the `cbprintf` code to support a different ABI when compiling for RV32E

Any input about this is welcome.

stephanosio, 2022-05-23:
From "The RISC-V Instruction Set Manual Volume 1: User-Level ISA **Version 2.1**", under "20.4 RV32E Calling Convention":

> RV32E uses a subset of the Soft-Float calling convention. As only 16 integer registers x0–x15 are
present, there are only six argument registers (x10–x15), two saved registers (x8–x9), and three
temporary registers (x5–x7). **The stack is kept aligned on a four-byte boundary.**

According to the above, the current behaviour is correct.

What baffles me is that this section was removed in the document Version 2.2 and on, and the stack alignment requirement for the RV32E seems to be no longer mentioned in the documentations (any idea what happened here?) ...

stephanosio, 2022-05-23:
See also https://github.com/riscv-non-isa/riscv-toolchain-conventions#specifying-stack-alignment:

> The default stack alignment is 16 bytes in RV32I and RV64I, and 4 bytes on RV32E.

stephanosio, 2022-05-23:
> the stack alignment requirement for the RV32E seems to be no longer mentioned in the documentations (any idea what happened here?) ...

To answer my own question, it looks like this section has been removed from the ISA manual since it is about the calling convention and there is a dedicated document for this.

This is now mentioned in the ABI documentation (see https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-cc.adoc#ilp32e-calling-convention):

> The ILP32E calling convention is designed to be usable with the RV32E ISA. This calling convention is the same as the integer calling convention, except for the following differences. **The stack pointer need only be aligned to a 32-bit boundary.**

In conclusion, the compiler behaviour is correct.

nordic-krch, 2022-05-23:
cbprintf relies on what is returned by `__alignof__`. Can you check what is returned for RV32E? For `qemu_riscv32` 8 is returned for `__alignof__(long long int)`.

carlocaione, 2022-05-23:
@nordic-krch it's `8` also for RV32E. Is that compatible with a stack alignment of `4` bytes?

carlocaione, 2022-05-23:
@nordic-krch proposed in chat a fix for the issue that works. I'll embed that in the PR.

closed 2022-05-23 by carlocaione