zephyrproject-rtos/sdk-ng #484: arm-zephyr-eabi-gdb: error while loading shared libraries: libpython3.8.so.1.0

oyvindronningstad, 2022-05-27:
When calling zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb I get the following error:

> arm-zephyr-eabi-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory

I'm using Ubuntu 22.04 which has Python 3.10.

stephanosio, 2022-05-27:
Please see https://github.com/zephyrproject-rtos/sdk-ng/issues/407#issuecomment-943880289

stephanosio, 2022-06-13:
Closing since this question has been answered.

Also note that, with https://github.com/zephyrproject-rtos/sdk-ng/issues/485, the default GDB executable is no longer linked to the libpython; instead, an alternate Python scripting-capable GDB variant (`gdb-py`) is provided.

closed 2022-06-13 by stephanosio