zephyrproject-rtos/sdk-ng #496: Re-enable ARC targets

stephanosio, 2022-06-14:
https://github.com/zephyrproject-rtos/sdk-ng/commit/3054f9216ef9f666204ab1a7f034743dc7a97584 has disabled the ARC targets because the patches for the latest GNU toolchains are not currently available (GCC patches are missing).

Re-enable the ARC targets once these patches are provided and merged.

This must be completed prior to releasing the Zephyr SDK 0.15.0.

stephanosio, 2022-06-30:
@abrodkin Do you have any updates on the GCC 12 patch for ARC? I am planning to do a beta release of 0.15.0, it would be great if you could provide one before then.

stephanosio, 2022-07-05:
@abrodkin ping

stephanosio, 2022-07-07:
> FYI, ARC64’s gcc12.1 should now be ready for zephyr SDK integration in `zephyr-gcc12` branch at FOSS for ARC github.

@ruuddw Thanks!

stephanosio, 2022-07-07:
It looks like [the newlib build is failing for some (I suppose) newly added ARC64 targets](https://github.com/zephyrproject-rtos/sdk-ng/runs/7239093121?check_suite_focus=true#step:13:307):

```
Error:     /__w/_temp/workspace/build/.build/arc64-zephyr-elf/src/newlib/newlib/libc/machine/arc64/memcpy.S:76: Error: opcode 'movl' not supported for target hs5x
```

@ruuddw @abrodkin @claziss Do you have any suggestions on how to fix this? I see that you guys have your own [newlib fork](https://github.com/foss-for-synopsys-dwc-arc-processors/newlib) as well, do you have any suggestions on which patch(es) to pull?

FYI, Zephyr SDK builds with https://github.com/zephyrproject-rtos/newlib-cygwin/tree/zephyr-newlib-3.3.0.

stephanosio, 2022-07-08:
Re: newlib build failure

I have cherry-picked some commits from the `arc64` branch and it seems to work. https://github.com/zephyrproject-rtos/newlib-cygwin/pull/6

ruuddw, 2022-07-08:
Thanks, I expect @abrodkin will pick this up next week again to see if anything was missed and do more testing.

closed 2022-07-08 by stephanosio