zephyrproject-rtos/sdk-ng #500: Add clang-format and clang-tidy utilities

stephanosio, 2022-06-17:
Include `clang-format` and `clang-tidy` linter in the Zephyr SDK because:

1. The distro-provided clang version may be outdated and not contain all the features required by the Zephyr linter configurations.
2. It is foreseeable that we may need to add some custom rules (to be eventually upstreamed) in order to support Zephyr-specific coding style.

jrobeson, 2022-06-21:
It also guarantees stable output from clang-format.  The same rules can have different output on different clang-format versions. Whether that's due to a bug, or is intentional, I'm not sure. I know it can be intentional in other ecosystems though.

stephanosio, 2022-07-05:
We have added the `clang-format` to the Python package requirements list so that the required version can be installed through the `pip` package manager (see https://github.com/zephyrproject-rtos/zephyr/pull/46951).

For now, I will hold off on implementing this to see how that approach works out.

stephanosio, 2022-11-26:
Closing since there is no immediate need for this.

We may consider this approach if we decide to maintain a fork of clang-format and clang-tidy with additional patches.

closed 2022-11-26 by stephanosio