zephyrproject-rtos/sdk-ng #510: Enabling code coverage on Zephyr SDK

stefanodellosa, 2022-07-01:
Hi,

I am performing an activity that requires to enable code coverage on Zephyr toolchain to actually show how much of SDK code (in particular gcc) is being stressed by Zephyr build process.

I did modify the configuration (see attachment) and build completes successfully but unfortunately no coverage files are produced. 

Could you please help to verify if my configuration is correct?

Thanks and Regards,
Stefano

[x86_64-zephyr-elf.zip](https://github.com/zephyrproject-rtos/sdk-ng/files/9029815/x86_64-zephyr-elf.zip)

stephanosio, 2022-07-01:
GCC provides `--enable-coverage` option during configure.

stefanodellosa, 2022-07-01:
Thanks for the info! Issue is how to properly embed such flag in crosstool-ng like configuration.. I did not find good documentation/reference to do so. 

Have you experience with that?

stephanosio, 2022-07-04:
> Thanks for the info! Issue is how to properly embed such flag in crosstool-ng like configuration

You should be able to use `CT_CC_GCC_EXTRA_CONFIG_ARRAY`:
https://github.com/zephyrproject-rtos/crosstool-ng/blob/b6b22fd4032b2ce3738087987024eae33ee7fc09/config/cc/gcc.in#L69