zephyrproject-rtos/sdk-ng #524: Evaluate feasibility of creating macOS universal binary

stephanosio, 2022-07-15:
macOS allows single executable to contain the images for multiple architectures (i.e. x86_64 and AArch64) -- this is known as a "universal binary."

Evaluate the feasibility of creating a Zephyr SDK distribution bundle containing "universal binaries" so that users do not need to manually choose the host platform type.

dberlin, 2022-09-04:
For the stuff that uses cmake, this should be as simple as using CMAKE_OSX_ARCHITECTURES=arm64;x86_64.

For everything else, assuming you can generate the individual arch binaries, the "lipo" tool can put them together for you.

The toolchains are the trickier part, and not on the distribution side - there are plenty of things that try to extract arches from the binary name, or assume that a single binary can't be two different toolchains, or ...

stephanosio, 2025-07-22:
Closing as not planned since this will increase the size of the already huge SDK tarball, and there is really no good reason to create "universal" macOS SDK tarball.