zephyrproject-rtos/sdk-ng #547: GCC emits GOT indirections for weak symbol references on AArch64

stephanosio, 2022-08-23:
Since the GCC 12.1 release (Zephyr SDK 0.15.0), GCC emits the "GOT indirections" for weak symbol references on AArch64, even when compiling with `-fno-pic` and `-fno-pie`.

Building [the following code](https://github.com/zephyrproject-rtos/zephyr/blob/e06359f61656d65ac49f245fe180bb59ffb4ee17/subsys/pm/pm.c#L114-L117):

```c
	extern __weak void
		pm_state_exit_post_ops(enum pm_state state, uint8_t substate_id);

	if (pm_state_exit_post_ops != NULL) {
```

The `pm_state_exit_post_ops`, a weak symbol, is referenced as follows:

_GCC 10.3 (Zephyr SDK 0.14.2)_

```
	adrp	x0, .LC0
	ldr	x0, [x0, #:lo12:.LC0]
	...
.LC0:
	.xword	pm_state_exit_post_ops
```

_GCC 12.1 (Zephyr SDK 0.15.0)_

```
	adrp	x0, :got:pm_state_exit_post_ops
	ldr	x0, [x0, :got_lo12:pm_state_exit_post_ops]
```

Note that the code emitted by GCC 12.1 makes use of `:got:`, which references the Global Offset Table (GOT). This effectively requires the `.got` section to be included in the final image ([these are currently discarded by the AArch64 linker script](https://github.com/zephyrproject-rtos/zephyr/blob/e7591615724894a2319ea2c09d7eb21b4263fbf0/include/zephyr/arch/arm64/scripts/linker.ld#L187-L193)).

**The Global Offset Table (GOT) should not be used when compiling position independent code (PIC).**

This issue was introduced by the GCC commit https://github.com/gcc-mirror/gcc/commit/fb0746f3a6b7fd0223efa71d0dc3fc02166e338b.

p.s. when the weak symbol is actually defined, the linker does populate the corresponding GOT entry with the address of the symbol at the default linking address (though, the "default" linking address does not really apply to the static executables ...).

stephanosio, 2022-08-23:
Related discussion on Discord: https://discord.com/channels/720317445772017664/733036650275537027/1011656252025212979

keith-packard, 2022-08-23:
If we can figure out how to get the GOT reduced in size, then this might end up a net win as multiple references to the same weak symbol could actually share GOT entries?

stephanosio, 2022-08-24:
https://github.com/zephyrproject-rtos/sdk-ng/pull/548 reverted the GCC commit causing this issue.

Note that this problem still exists in the upstream GCC and its users will still be affected by it.

closed 2022-08-24 by stephanosio