zephyrproject-rtos/sdk-ng #556: arm-zephyr-eabi-gdb crashing on using blackmagicprobe as the runner.

jo5huajohn, 2022-09-04:
**Describe the bug**
On running west flash -r blackmagicprobe, arm-zephyr-eabi-gdb crashes with:internal-error: inferior_thread: Assertion `current_thread_ != nullptr' failed.

Please also mention any information which could help others to understand
the problem you're facing:
 - **Target Platform Used:** adafruit_feather_nrf52840
 - **What have you tried to diagnose or workaround this issue?**
 Downgraded to sdk 0.14.2
 - **Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?**
Issue was not observed in sdk 0.14.2. I have not used git bisect to identify the specific commit.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to zephyr/samples/basic/blinky.
2. Start a build using west build -p auto -b adafruit_feather_nrf52840.
3. Flash it using west flash -r blackmagicprobe.
4. See error:Remote debugging using /dev/ttyBmpGdb
> Target voltage: 0.00V
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
Attaching to Remote target
/\__w/_temp/workspace/build/.build/arm-zephyr-eabi/src/gdb/gdb/thread.c:85: internal-error: inferior_thread: Assertion `current_thread_ != nullptr' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
----- Backtrace -----
0x55e69136785d ???
0x55e69165981c ???
0x55e6916599fa ???
0x55e69174738e ???
0x55e6912a0378 ???
0x55e69160ea86 ???
0x55e6915119c5 ???
0x55e6914c67ed ???
0x55e6916105f4 ???
0x55e691577096 ???
0x55e691581f2a ???
0x55e6914aa38b ???
0x55e69139908f ???
0x55e691618f9b ???
0x55e6914f12d1 ???
0x55e6914f13f0 ???
0x55e6914f2a74 ???
0x55e6914f362a ???
0x55e6912a246a ???
0x7f9c9ba29d8f \__libc_start_call_main
	../sysdeps/nptl/libc_start_call_main.h:58
0x7f9c9ba29e3f \__libc_start_main_impl
	../csu/libc-start.c:392
0x55e6912b2e79 ???
0xffffffffffffffff ???
\---------------------
>
> This is a bug, please report it.  For instructions, see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
>
> FATAL ERROR: command exited with status -6: /home/joshua/.local/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb -ex 'set confirm off' -ex 'target extended-remote /dev/ttyBmpGdb' -ex 'monitor swdp_scan' -ex 'attach 1' -ex 'load /home/joshua/Workspaces/zephyr_workspace/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf' -ex kill -ex quit -silent
>

**Expected behavior**
Image flash should be successful.

arbrauns, 2022-10-14:
Upstream GDB issue, see blackmagic-debug/blackmagic#929 for details.

stephanosio, 2022-11-11:
A workaround to make Blackmagic GDB server work with GDB 12 (included in Zephyr SDK 0.15.0 and above) has been added in the v1.8.2 release. For more details, see https://github.com/blackmagic-debug/blackmagic/releases/tag/v1.8.2

Closing this issue since a workaround has been implemented and Zephyr SDK users should not be seeing this issue anymore.

closed 2022-11-11 by stephanosio