zephyrproject-rtos/sdk-ng #558: stm32u5xx_hal_dma_ex.c:4719:1: internal compiler error: Segmentation fault

ext-mwo, 2022-09-14:
I get a crash while building Zaphyr/TF-M.

The GCC version:
```
ext-mwo@extmwo-ThinkPad-T14-Gen-2i:~/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700$ /home/ext-mwo/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=/home/ext-mwo/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
COLLECT_LTO_WRAPPER=/home/ext-mwo/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../libexec/gcc/arm-zephyr-eabi/12.1.0/lto-wrapper
Target: arm-zephyr-eabi
Configured with: /__w/_temp/workspace/build/.build/arm-zephyr-eabi/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=arm-zephyr-eabi --prefix=/__w/_temp/workspace/output/arm-zephyr-eabi --exec_prefix=/__w/_temp/workspace/output/arm-zephyr-eabi --with-local-prefix=/__w/_temp/workspace/output/arm-zephyr-eabi/arm-zephyr-eabi --with-headers=/__w/_temp/workspace/output/arm-zephyr-eabi/arm-zephyr-eabi/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='Zephyr SDK 0.15.0' --with-bugurl=https://github.com/zephyrproject-rtos/sdk-ng/issues --enable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-libstdcxx-verbose --with-gmp=/__w/_temp/workspace/build/.build/arm-zephyr-eabi/buildtools --with-mpfr=/__w/_temp/workspace/build/.build/arm-zephyr-eabi/buildtools --with-mpc=/__w/_temp/workspace/build/.build/arm-zephyr-eabi/buildtools --with-isl=/__w/_temp/workspace/build/.build/arm-zephyr-eabi/buildtools --enable-lto --enable-target-optspace --disable-nls --enable-multiarch --with-multilib-list=rmprofile --enable-languages=c,c++ --with-gnu-ld --with-gnu-as --enable-initfini-array
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.1.0 (Zephyr SDK 0.15.0) 

```
Error message:
```
during GIMPLE pass: evrp
/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ReplaceNode_Head':
/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c:4719:1: internal compiler error: Segmentation fault
 4719 | }
      | ^
0x7f896be5751f ???
	./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0
0x7f896be3ed8f __libc_start_call_main
	../sysdeps/nptl/libc_start_call_main.h:58
0x7f896be3ee3f __libc_start_main_impl
	../csu/libc-start.c:392
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://github.com/zephyrproject-rtos/sdk-ng/issues> for instructions.
```
The command line was:
```
/home/ext-mwo/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBL1 -DBL1_HEADER_SIZE=0x000 -DBL1_TRAILER_SIZE=0x000 -DBL2 -DBL2_HEADER_SIZE=0x400 -DBL2_TRAILER_SIZE=0x2000 -DCONFIG_TFM_BUILDING_SPE=1 -DCONFIG_TFM_DOORBELL_API=1 -DCONFIG_TFM_FP=0 -DCONFIG_TFM_PARTITION_META -DCRYPTO_HW_ACCELERATOR -DDAUTH_CHIP_DEFAULT -DMBEDTLS_CONFIG_FILE=\"/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/lib/ext/mbedcrypto/mbedcrypto_config/tfm_mbedcrypto_config_default.h\" -DMBEDTLS_GCM_C -DMCUBOOT_IMAGE_NUMBER=2 -DMCUBOOT_PRIMARY_ONLY -DMCUBOOT_SIGN_RSA -DMCUBOOT_SIGN_RSA_LEN=3072 -DPLATFORM_DEFAULT_NV_SEED -DPLATFORM_DEFAULT_OTP -DPSA_CRYPTO_SECURE -DSTM32U585xx -DTFM_FIH_PROFILE_OFF -DTFM_LVL=1 -DTFM_PARTITION_CRYPTO -DTFM_PARTITION_INITIAL_ATTESTATION -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_PARTITION_PLATFORM -DTFM_PARTITION_PROTECTED_STORAGE -DTFM_PSA_API -DTFM_SPM_LOG_LEVEL=TFM_SPM_LOG_LEVEL_INFO -DTFM_SPM_LOG_RAW_ENABLED -DTFM_SP_LOG_RAW_ENABLED -DUSE_HAL_DRIVER -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/common/stm32u5xx -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/b_doam0/../common/hal/Native_Driver -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/b_doam0/../common/hal/CMSIS_Driver -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/common/stm32u5xx/Device/Include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/common/stm32u5xx/hal/Inc -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/common/stm32u5xx/secure -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/crypto/mbedtls/library -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/accelerator/interface -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/./ext -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/./ext/cmsis -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/./ext/common -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/./ext/driver -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/./include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/b_doam0/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/tu/common/stm32u5xx/boards -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/lib/fih/inc -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/interface/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/build/tfm/generated/interface/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/spm/cmsis_psa -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/spm/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/spm/include/interface -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/spm/arch/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/spm/cmsis_psa/arch -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/spm -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/spm/cmsis_psa/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/partitions/lib/sprt/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/crypto/mbedtls/include -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/modules/tee/tf-m/trusted-firmware-m/secure_fw/partitions/crypto -I/home/ext-mwo/devel/remote-git/griffin-dev-tu700-new/griffin-dev-tu700/zephyr_base/mo [... cut ...]
The command doesn't generate a core file - I enabled core generation in the current directory.

So I attach strace output and Src.tar.gz which contains the files generated when adding "-v -save-temps" to the command line above.

[process_dump.10635.txt](https://github.com/zephyrproject-rtos/sdk-ng/files/9566635/process_dump.10635.txt)
[process_dump.10636.txt](https://github.com/zephyrproject-rtos/sdk-ng/files/9566636/process_dump.10636.txt)
[Src.tar.gz](https://github.com/zephyrproject-rtos/sdk-ng/files/9566648/Src.tar.gz)

theotherjimmy, 2022-09-14:
where did you look for the core file? With systemd, you can easily use coredumpctl to get at it, and it's stored is a strangely-named directory in `/var`. Here's another issue with the same error:
https://github.com/nvim-treesitter/nvim-treesitter/issues/2195

theotherjimmy, 2022-09-14:
EVRP is a value range propagation optimization

ext-mwo, 2022-09-15:
> where did you look for the core file? With systemd, you can easily use coredumpctl to get at it, and it's stored is a strangely-named directory in `/var`. Here's another issue with the same error: [nvim-treesitter/nvim-treesitter#2195](https://github.com/nvim-treesitter/nvim-treesitter/issues/2195)

I spent some time on making coredumps work and made it create them in the current directory. On ubuntu they are placed in /var/crash or some such directory.

I executed sleep 10 and killed it with some flags - got a coredump in the same directory so the coredump functionality of my OS works. :)

This is why this is so strange and why I attached the output from strace for both gcc and the child process it creates.

Rahul170809, 2023-11-02:
I am building project for STM32U575 using default tool chain STMCubeIDE (which is `**arm-none-eabi-gcc-11-3-rel1**`), my code is getting compile without any error. However, when I tried to compile the same using the tool chain `**arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi**` downloaded from [ARM GNU Toolchain Downloads](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) is giving the same error as mentioned in this issue.

Could anyone please help me to fix it ?

Thank you

SeanMollet, 2024-10-05:
@Rahul170809 I upgraded to the latest arm-none-eabi-gcc (13.3.1) and it builds correctly now.

nashif, 2025-05-03:
is this still an issue?