zephyrproject-rtos/sdk-ng #563: Zephyr build using Zephyr sdk 0.14.2 taking more time

ShubhadaHarkut-eaton, 2022-09-29:
When trying to build the zephyr blinky example code using the sdk 0.14.2 it take more time to complete the build as compared to armgnu tool

stephanosio, 2022-09-29:
can you provide some numbers to compare and information on your environment (OS, version, ...).

ShubhadaHarkut-eaton, 2022-09-29:
I am using zephyr version 3.1.0 with sdk version 0.14.2 
and to compare with zephyr version 3.1.0 with GNUArmEmbeddedToolchain 10 2021.10

stephanosio, 2022-09-29:
Please provide:

1. your host OS type and version
1. the compilation time difference observed
    a) time taken with GNU Arm Embedded 2021.10
    b) time taken with Zephyr SDK 0.14.2

stephanosio, 2022-09-29:
## Set-up

Ubuntu 20.04 x86_64
Zephyr tag v3.1.0

```
$ pushd samples/basic/blinky
$ mkdir build && cd build
```

for each run:

```
$ rm -rf *
$ ccache -C
$ cmake -GNinja -DBOARD=nrf52840dk_nrf52840 ..
$ time ninja
```

## Zephyr SDK 0.14.2

real    0m5.920s
user    0m45.413s
sys     0m16.816s

real    0m5.420s
user    0m45.224s
sys     0m16.895s

real    0m5.885s
user    0m44.881s
sys     0m16.849s

## GNU Arm Embedded 2021.10

real    0m5.456s
user    0m43.509s
sys     0m14.652s

real    0m5.867s
user    0m44.239s
sys     0m15.435s

real    0m5.712s
user    0m43.714s
sys     0m16.066s

## Conclusion

Zephyr SDK 0.14.2        avg(real)  0m5.742s
GNU Arm Embedded 2021.10 avg(real)  0m5.678s

No discernible differences have been found.

closed 2022-09-29 by stephanosio