zephyrproject-rtos/sdk-ng #566: _GLIBCXX_USE_C99_MATH_TR1 is undefined for some architectures

stephanosio, 2022-10-02:
`_GLIBCXX_USE_C99_MATH_TR1` is undefined for some architectures (e.g. in `aarch64-zephyr-elf` toolchain), leading to build issues with `cmath` such as:

```
/Users/cfriedt/workspace/gsoc-2022-thrift/.upstream/lib/cpp/src/thrift/transport/TBufferTransports.cpp: In member function 'void apache::thrift::transport::TMemoryBuffer::ensureCanWrite(uint32_t)':
/Users/cfriedt/workspace/gsoc-2022-thrift/.upstream/lib/cpp/src/thrift/transport/TBufferTransports.cpp:374:45: error: 'exp2' is not a member of 'std'
  374 |   const double suggested_buffer_size = std::exp2(std::ceil(std::log2(required_buffer_size)));
      |                                             ^~~~
/Users/cfriedt/workspace/gsoc-2022-thrift/.upstream/lib/cpp/src/thrift/transport/TBufferTransports.cpp:374:55: error: 'ceil' is not a member of 'std'
  374 |   const double suggested_buffer_size = std::exp2(std::ceil(std::log2(required_buffer_size)));
      |                                                       ^~~~
/Users/cfriedt/workspace/gsoc-2022-thrift/.upstream/lib/cpp/src/thrift/transport/TBufferTransports.cpp:374:65: error: 'log2' is not a member of 'std'
  374 |   const double suggested_buffer_size = std::exp2(std::ceil(std::log2(required_buffer_size)));
```

cfriedt, 2022-10-27:
This appears to be impacting at least rv32, rv64 (`riscv64-zephyr-elf` toolchain), x86, x86_64 (`x86_64-zephyr-elf` toolchain), and aarch64 (`aarch64-zephyr-elf`).

The only "popular" architecture that is not affected by this is, from what I can tell, is arm (`arm-zephyr-eabi` toolchain)

stephanosio, 2022-11-15:
List of the targets that do not define `_GLIBCXX_USE_C99_MATH_TR1` as of Zephyr SDK 0.15.1:

* aarch64-zephyr-elf
* riscv64-zephyr-elf
* x86_64-zephyr-elf

As expected, the test code used by the autoconf check does fail to compile on the above targets.

**aarch64-zephyr-elf**

```
mathtest.c:10:9: error: 'acoshl' was not declared in this scope; did you mean 'acoshf'?
   10 |         acoshl(0.0l);
      |         ^~~~~~
      |         acoshf
mathtest.c:13:9: error: 'asinhl' was not declared in this scope; did you mean 'asinhf'?
   13 |         asinhl(0.0l);
      |         ^~~~~~
      |         asinhf
mathtest.c:16:9: error: 'atanhl' was not declared in this scope; did you mean 'atanhf'?
   16 |         atanhl(0.0l);
      |         ^~~~~~
      |         atanhf
mathtest.c:19:9: error: 'cbrtl' was not declared in this scope; did you mean 'cbrtf'?
   19 |         cbrtl(0.0l);
      |         ^~~~~
      |         cbrtf
mathtest.c:22:9: error: 'copysignl' was not declared in this scope; did you mean 'copysignf'?
   22 |         copysignl(0.0l, 0.0l);
      |         ^~~~~~~~~
      |         copysignf
mathtest.c:25:9: error: 'erfl' was not declared in this scope; did you mean 'erff'?
   25 |         erfl(0.0l);
      |         ^~~~
      |         erff
mathtest.c:28:9: error: 'erfcl' was not declared in this scope; did you mean 'erfcf'?
   28 |         erfcl(0.0l);
      |         ^~~~~
      |         erfcf
mathtest.c:31:9: error: 'exp2l' was not declared in this scope; did you mean 'exp2f'?
   31 |         exp2l(0.0l);
      |         ^~~~~
      |         exp2f
mathtest.c:34:9: error: 'expm1l' was not declared in this scope; did you mean 'expm1f'?
   34 |         expm1l(0.0l);
      |         ^~~~~~
      |         expm1f
mathtest.c:37:9: error: 'fdiml' was not declared in this scope; did you mean 'fdimf'?
   37 |         fdiml(0.0l, 0.0l);
      |         ^~~~~
      |         fdimf
mathtest.c:40:9: error: 'fmal' was not declared in this scope; did you mean 'fmax'?
   40 |         fmal(0.0l, 0.0l, 0.0l);
      |         ^~~~
      |         fmax
mathtest.c:43:9: error: 'fmaxl' was not declared in this scope; did you mean 'fmaxf'?
   43 |         fmaxl(0.0l, 0.0l);
      |         ^~~~~
      |         fmaxf
mathtest.c:46:9: error: 'fminl' was not declared in this scope; did you mean 'fminf'?
   46 |         fminl(0.0l, 0.0l);
      |         ^~~~~
      |         fminf
mathtest.c:52:9: error: 'ilogbl' was not declared in this scope; did you mean 'ilogbf'?
   52 |         ilogbl(0.0l);
      |         ^~~~~~
      |         ilogbf
mathtest.c:55:9: error: 'lgammal' was not declared in this scope; did you mean 'lgammaf'?
   55 |         lgammal(0.0l);
      |         ^~~~~~~
[... 350 lines cut ...]
mathtest.c:99:9: error: 'rintl' was not declared in this scope; did you mean 'rintf'?
   99 |         rintl(0.0l);
      |         ^~~~~
      |         rintf
mathtest.c:102:9: error: 'roundl' was not declared in this scope; did you mean 'roundf'?
  102 |         roundl(0.0l);
      |         ^~~~~~
      |         roundf
mathtest.c:105:9: error: 'scalblnl' was not declared in this scope; did you mean 'scalblnf'?
  105 |         scalblnl(0.0l, 0l);
      |         ^~~~~~~~
      |         scalblnf
mathtest.c:108:9: error: 'scalbnl' was not declared in this scope; did you mean 'scalbnf'?
  108 |         scalbnl(0.0l, 0);
      |         ^~~~~~~
      |         scalbnf
mathtest.c:111:9: error: 'tgammal' was not declared in this scope; did you mean 'tgammaf'?
  111 |         tgammal(0.0l);
      |         ^~~~~~~
      |         tgammaf
mathtest.c:114:9: error: 'truncl' was not declared in this scope; did you mean 'truncf'?
  114 |         truncl(0.0l);
      |         ^~~~~~
      |         truncf
```

stephanosio, 2022-11-15:
Newlib does not provide the `long double` math functions (i.e. the math functions suffixed with `l`) for these targets because:

* Newlib does not implement any generic `long double` math functions.
* Newlib maps the `long double` math functions to the `double` math functions if and only if the type of `long double` is equivalent to the type of `double`.

On the **unaffected** targets, `typeof(long double) == typeof(double)`, so the `long double` math functions are mapped to the `double` math functions.

On the **affected** targets, however, `typeof(long double) != typeof(double)`, so the `long double` math functions are NOT mapped to the `double` math functions, and none of the affected targets provide target-specific `long double` math functions; leaving the `long double` math functions unimplemented.

In summary, `typeof(long double) != typeof(double)` on the affected targets, which leads to newlib not providing the `long double` math functions on them, which leads to the libstdc++ not defining `_GLIBCXX_USE_C99_MATH_TR1` because the check for it expects the `long double` math functions to be available, which eventually leads to some math functions not being declared by the `cmath` header.

Now the question is:

1. Why is `typeof(long double) != typeof(double)` on the affected targets and should that be the case?
2. Should libstdc++ require the `long double` math functions as part of the check for defining `_GLIBCXX_USE_C99_MATH_TR1`?

stephanosio, 2022-11-15:
> 1\. Why is `typeof(long double) != typeof(double)` on the affected targets and should that be the case?

For x86_64-zephyr-elf, the answer is obvious -- because x86(-64) FPU supports 80-bit floating point type; however, the `sizeof(long double)` is not 10 or 12 (which is a common convention in x86-32 code). Instead, `sizeof(long double) == 16` -- [the rationale is given in the GCC documentation](https://gcc.gnu.org/onlinedocs/gcc/x86-Options.html):

> In the x86-64 compiler, -m128bit-long-double is the default choice as its ABI specifies that long double is aligned on 16-byte boundary.

For aarch64-zephyr-elf, [the AAPCS64 defines `long double` to be "quad precision (IEEE 754-2008)."](https://github.com/ARM-software/abi-aa/blob/main/aapcs64/aapcs64.rst)

For riscv64-zephyr-elf, [the RISC-V ELF psABI defines `long double` as a 16-byte data type.](https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-cc.adoc)

In conclusion, `typeof(long double) != typeof(double)` makes sense for these targets and overriding `typeof(long double)` to be `typeof(dobule)` will be in violation of the ABI.

cfriedt, 2022-11-15:
I was a bit surprised but apparently long double is part of C99.

> Now the question is:
> 
> 1. Why is `typeof(long double) != typeof(double)` on the affected targets and should that be the case?

Some architectures do have e.g. 80-bit, 128-bit hardware floating point support and yeah, it's ok for them to define those types.

> 2. Should libstdc++ require the `long double` math functions as part of the check for defining `_GLIBCXX_USE_C99_MATH_TR1`?

It's quite possible that it should given that long double is part of C99.

It's probably safe to simply wrap the double math functions to provide long double math functions for the time being if they're completely absent.

I don't believe they can be aliases because a cast is necessary if the underlying type actually differs.

stephanosio, 2022-11-15:
Re: https://github.com/zephyrproject-rtos/sdk-ng/issues/566#issuecomment-1315165640

> 2\. Should libstdc++ require the `long double` math functions as part of the check for defining `_GLIBCXX_USE_C99_MATH_TR1`?

The `_GLIBCXX_USE_C99_MATH_TR1` definition is used in the libstdc++ headers to [determine whether the C++ TR1 math functions should be "imported":](https://github.com/zephyrproject-rtos/gcc/blob/97cb645261c5100096ed09bcbba6ad10d540c5e5/libstdc++-v3/include/c_global/cmath#L954-L1844)

```c++
...
  // functions
  using ::acosh;
  using ::acoshf;
  using ::acoshl;
...
```

Since the C++ compiler will emit an error when, for instance, `acoshl` is not declared, the test for defining `_GLIBCXX_USE_C99_MATH_TR1` should indeed check for the availability of the `long double` functions.

The behaviour of libstdc++ is correct because the C++ TR1 requires the `long double` math functions such as `acoshl` in the `cmath` header (see `tr.c99.cmath.syn`) -- the underlying issue here is that newlib does not provide the `long double` math functions, which is not in compliance with the C99 standard that lists these functions under the `math.h`.

In that sense, the "correct" way of fixing this would be to add the `long double` math support to the newlib in order to achieve feature parity across all targets; but, that is by no means a small task and will require extensive development and testing from our side.

**An intermediary solution, until we achieve `long double` feature parity across all targets, would be to make the `long double` math function support an optional feature in the libstdc++.** This can be relatively easily achieved by adding some additional autoconf checks and ifdefs.

In summary, we will be implementing a libstdc++ patch to make the `long double` math function support in the C++ TR1 an optional feature for now, with the goal of eventually adding the `long double` math function support across all supported targets in the libc.

Also note that the picolibc suffers from the same problem (FYI @keith-packard).

stephanosio, 2022-11-15:
> In summary, we will be implementing a libstdc++ patch to make the `long double` math function support in the C++ TR1 an optional feature for now, with the goal of eventually adding the `long double` math function support across all supported targets in the libc.

Created #593 to track this.

keith-packard, 2022-11-15:
Yup, picolibc also lacks support for many math functions for floats other than 32/64 IEEE. I'm not entirely sure how to manage this though -- there are at least three "standards" for larger values: 80-bit intel, 128-bit ieee and IBM "double-double". Each should have custom code written for it. I think we could approach this incrementally, starting with the more common functions and perhaps putting off things like bessel and gamma functions for some future date. I'm not sure who might be persuaded to do this work though; it's pretty low value for most projects as few embedded systems actually have hardware for larger floating point values.

cfriedt, 2022-11-15:
> In summary, we will be implementing a libstdc++ patch to make the `long double` math function support in the C++ TR1 an optional feature for now, with the goal of eventually adding the `long double` math function support across all supported targets in the libc.

Seems reasonable to me 👍

keith-packard, 2022-11-16:
> I'm not sure who might be persuaded to do this work though; it's pretty low value for most projects as few embedded systems actually have hardware for larger floating point values.

oops. https://github.com/picolibc/picolibc/pull/378

cfriedt, 2022-11-16:
> > I'm not sure who might be persuaded to do this work though; it's pretty low value for most projects as few embedded systems actually have hardware for larger floating point values.
> 
> oops. https://github.com/picolibc/picolibc/pull/378

How??

keith-packard, 2022-11-17:
> How??

Mistakes were made? Hoping to finish the math library cleanups that are ahead of this in another day, then this should land without issue (there isn't much testing aside from validating that the entry points exist).

PR #594 (2022-11-17): gcc: Pull in libstdc++ long double math function support fix
This commit pulls in the GCC commit that makes the long double math function support in the libstdc++ <cmath> header an optional feature such that the C99 math functions are available in the header when libstdc++ is compiled with a C library that does not provide the long double math functions  (e.g. newlib).

GCC PR: https://github.com/zephyrproject-rtos/gcc/pull/18

Fixes #566
closed 2022-11-17 by stephanosio