zephyrproject-rtos/sdk-ng #575: Consistent target naming suffixes

beriberikix, 2022-10-09:
Currently, most targets end with `-zephyr-elf`, while the xtensa-based ones end with `_zephyr-elf`. It makes automation a little more challenging.

```
  - aarch64-zephyr-elf
  - arc64-zephyr-elf
  - arc-zephyr-elf
  - arm-zephyr-eabi
  - mips-zephyr-elf
  - nios2-zephyr-elf
  - riscv64-zephyr-elf
  - sparc-zephyr-elf
  - x86_64-zephyr-elf
  - xtensa-espressif_esp32_zephyr-elf
  - xtensa-espressif_esp32s2_zephyr-elf
  - xtensa-intel_apl_adsp_zephyr-elf
  - xtensa-intel_bdw_adsp_zephyr-elf
  - xtensa-intel_byt_adsp_zephyr-elf
  - xtensa-intel_s1000_zephyr-elf
  - xtensa-nxp_imx_adsp_zephyr-elf
  - xtensa-nxp_imx8m_adsp_zephyr-elf
  - xtensa-sample_controller_zephyr-elf
```

beriberikix, 2022-10-09:
Also arm is called `arm-zephyr-eabi` :)

stephanosio, 2022-10-09:
This is just how GNU toolchain target triplet works and we cannot change the current naming scheme without significantly modifying the toolchain component build scripts.

It should get much better once #350 is implemented, however.

beriberikix, 2022-10-09:
Roger! Will close :)

closed 2022-10-09 by beriberikix