zephyrproject-rtos/sdk-ng #576: GDB doesn't work with "custom" $TERM

tyalie, 2022-10-12:
I'm using the kitty terminal which comes with the `xterm-kitty` term. When using it (or `xterm-utf8`, ...) I've similar behaviour to setting `TERM=unknown` in that backspace and other keys don't properly work anymore. I can fix this by manually changing the term to e.g. `xterm`. I don't have that issue when using other debuggers like `gdb` or `gdb-multiarch`.

## Environment
The term is installed in the appropriate directory in `/usr/share/terminfo`

My default environment variables 
```
$ env | grep TERM
TERM=xterm-kitty
COLORTERM=truecolor
TERMINFO=/opt/kitty.app/lib/kitty/terminfo
```

And the `infocmp` dump for `xterm` and `xterm-kitty`
<details>
  <summary><code>infocmp -a xterm-kitty</code></summary>
<pre><code>#	Reconstructed via infocmp from file: /opt/kitty.app/lib/kitty/terminfo/x/xterm-kitty
xterm-kitty|KovIdTTY,
	am, ccc, hs, km, mc5i, mir, msgr, npc, xenl, Su, Tc, fullkbd,
	colors#0x100, cols#80, it#8, lines#24, pairs#0x7fff,
	acsc=++\,\,--..00``aaffgghhiijjkkllmmnnooppqqrrssttuuvvwwxxyyzz{{||}}~~,
	bel=^G, bold=\E[1m, cbt=\E[Z, civis=\E[?25l,
	clear=\E[H\E[2J, cnorm=\E[?12h\E[?25h, cr=\r,
	csr=\E[%i%p1%d;%p2%dr, cub=\E[%p1%dD, cub1=^H,
	cud=\E[%p1%dB, cud1=\n, cuf=\E[%p1%dC, cuf1=\E[C,
	cup=\E[%i%p1%d;%p2%dH, cuu=\E[%p1%dA, cuu1=\E[A,
	cvvis=\E[?12;25h, dch=\E[%p1%dP, dch1=\E[P, dim=\E[2m,
	dl=\E[%p1%dM, dl1=\E[M, dsl=\E]2;\007, ech=\E[%p1%dX,
	ed=\E[J, el=\E[K, el1=\E[1K, flash=\E[?5h$<100/>\E[?5l,
	fsl=^G, home=\E[H, hpa=\E[%i%p1%dG, ht=^I, hts=\EH,
	ich=\E[%p1%d@, il=\E[%p1%dL, il1=\E[L, ind=\n,
	indn=\E[%p1%dS,
	initc=\E]4;%p1%d;rgb:%p2%{255}%*%{1000}%/%2.2X/%p3%{255}%*%{1000}%/%2.2X/%p4%{255}%*%{1000}%/%2.2X\E\\,
	kBEG=\E[1;2E, kDC=\E[3;2~, kEND=\E[1;2F, kHOM=\E[1;2H,
	kIC=\E[2;2~, kLFT=\E[1;2D, kNXT=\E[6;2~, kPRV=\E[5;2~,
	kRIT=\E[1;2C, ka1=, ka3=, kbeg=\EOE, kbs=^?, kc1=, kc3=,
	kcbt=\E[Z, kcub1=\EOD, kcud1=\EOB, kcuf1=\EOC, kcuu1=\EOA,
	kdch1=\E[3~, kend=\EOF, kf1=\EOP, kf10=\E[21~, kf11=\E[23~,
	kf12=\E[24~, kf13=\E[1;2P, kf14=\E[1;2Q, kf15=\E[1;2R,
	kf16=\E[1;2S, kf17=\E[15;2~, kf18=\E[17;2~,
	kf19=\E[18;2~, kf2=\EOQ, kf20=\E[19;2~, kf21=\E[20;2~,
	kf22=\E[21;2~, kf23=\E[23;2~, kf24=\E[24;2~,
	kf25=\E[1;5P, kf26=\E[1;5Q, kf27=\E[1;5R, kf28=\E[1;5S,
	kf29=\E[15;5~, kf3=\EOR, kf30=\E[17;5~, kf31=\E[18;5~,
	kf32=\E[19;5~, kf33=\E[20;5~, kf34=\E[21;5~,
	kf35=\E[23;5~, kf36=\E[24;5~, kf37=\E[1;6P, kf38=\E[1;6Q,
	kf39=\E[1;6R, kf4=\EOS, kf40=\E[1;6S, kf41=\E[15;6~,
	kf42=\E[17;6~, kf43=\E[18;6~, kf44=\E[19;6~,
	kf45=\E[20;6~, kf46=\E[21;6~, kf47=\E[23;6~,
	kf48=\E[24;6~, kf49=\E[1;3P, kf5=\E[15~, kf50=\E[1;3Q,
	kf51=\E[1;3R, kf52=\E[1;3S, kf53=\E[15;3~, kf54=\E[17;3~,
	kf55=\E[18;3~, kf56=\E[19;3~, kf57=\E[20;3~,
	kf58=\E[21;3~, kf59=\E[23;3~, kf6=\E[17~, kf60=\E[24;3~,
	kf61=\E[1;4P, kf62=\E[1;4Q, kf63=\E[1;4R, kf7=\E[18~,
	kf8=\E[19~, kf9=\E[20~, khlp=, khome=\EOH, kich1=\E[2~,
	kind=\E[1;2B, kmous=\E[M, knp=\E[6~, kpp=\E[5~,
	kri=\E[1;2A, kund=, oc=\E]104\007, op=\E[39;49m, rc=\E8,
	rep=%p1%c\E[%p2%{1}%-%db, rev=\E[7m, ri=\EM,
	rin=\E[%p1%dT, ritm=\E[23m, rmacs=\E(B, rmam=\E[?7l,
	rmcup=\E[?1049l, rmir=\E[4l, rmkx=\E[?1l, rmso=\E[27m,
	rmul=\E[24m, rs1=\E]\E\\\Ec, sc=\E7,
	setab=\E[%?%p1%{8}%<%t4%p1%d%e%p1%{16}%<%t10%p1%{8}%-%d%e48;5;%p1%d%;m,
	setaf=\E[%?%p1%{8}%<%t3%p1%d%e%p1%{16}%<%t9%p1%{8}%-%d%e38;5;%p1%d%;m,
	sgr=%?%p9%t\E(0%e\E(B%;\E[0%?%p6%t;1%;%?%p2%t;4%;%?%p1%p3%|%t;7%;%?%p4%t;5%;%?%p7%t;8%;m,
	sgr0=\E(B\E[m, sitm=\E[3m, smacs=\E(0, smam=\E[?7h,
	smcup=\E[?1049h, smir=\E[4h, smkx=\E[?1h, smso=\E[7m,
	smul=\E[4m, tbc=\E[3g, tsl=\E]2;, u6=\E[%i%d;%dR, u7=\E[6n,
	u8=\E[?%[;0123456789]c, u9=\E[c, vpa=\E[%i%p1%dd,
	Smulx=\E[4:%p1%dm, Sync=\EP=%p1%ds\E\\, kBEG3=\E[1;3E,
	kBEG4=\E[1;4E, kBEG5=\E[1;5E, kBEG6=\E[1;6E,
	kBEG7=\E[1;7E, kDC3=\E[3;3~, kDC4=\E[3;4~, kDC5=\E[3;5~,
	kDC6=\E[3;6~, kDC7=\E[3;7~, kDN=\E[1;2B, kDN3=\E[1;3B,
	kDN4=\E[1;4B, kDN5=\E[1;5B, kDN6=\E[1;6B, kDN7=\E[1;7B,
	kEND3=\E[1;3F, kEND4=\E[1;4F, kEND5=\E[1;5F,
	kEND6=\E[1;6F, kEND7=\E[1;7F, kHOM3=\E[1;3H,
	kHOM4=\E[1;4H, kHOM5=\E[1;5H, kHOM6=\E[1;6H,
	kHOM7=\E[1;7H, kIC3=\E[2;3~, kIC4=\E[2;4~, kIC5=\E[2;5~,
	kIC6=\E[2;6~, kIC7=\E[2;7~, kLFT3=\E[1;3D, kLFT4=\E[1;4D,
	kLFT5=\E[1;5D, kLFT6=\E[1;6D, kLFT7=\E[1;7D,
	kNXT3=\E[6;3~, kNXT4=\E[6;4~, kNXT5=\E[6;5~,
	kNXT6=\E[6;6~, kNXT7=\E[6;7~, kPRV3=\E[5;3~,
	kPRV4=\E[5;4~, kPRV5=\E[5;5~, kPRV6=\E[5;6~,
	kPRV7=\E[5;7~, kRIT3=\E[1;3C, kRIT4=\E[1;4C,
	kRIT5=\E[1;5C, kRIT6=\E[1;6C, kRIT7=\E[1;7C, kUP=\E[1;2A,
	kUP3=\E[1;3A, kUP4=\E[1;4A, kUP5=\E[1;5A, kUP6=\E[1;6A,
	kUP7=\E[1;7A, rmxx=\E[29m,
	setrgbb=\E[48:2:%p1%d:%p2%d:%p3%dm,
	setrgbf=\E[38:2:%p1%d:%p2%d:%p3%dm, smxx=\E[9m,</code></pre>
</details>

tyalie, 2022-10-12:
I can only guess on the reasons, but it feels like that this distribution of gdb doesn't correctly parse the terminfo entries.

stephanosio, 2022-10-12:
Zephyr toolchains do not link to the host ncurses library for compatibility reasons.

They statically link to a custom ncurses library built with the embedded support for common terminal types such as xterm without relying on the host terminfo DB, once again, for compatibility/distro agnosticity reasons.

tyalie, 2022-10-12:
okay interesting. Is there any workaround thoo?

keith-packard, 2022-10-12:
> okay interesting. Is there any workaround thoo?

You can build your own version of gdb that uses the local ncurses library.

stephanosio, 2022-10-12:
If your terminal emulator (kitty?) supports a more conventional mode (I mean it really should), then try that.

The following terminal types are supported by the Zephyr toolchains:
`linux,xterm,xterm-color,xterm-256color,vt100`

stephanosio, 2022-10-25:
Closing as a known issue since the limitation is a deliberate one for cross-distro compatibility.

closed 2022-10-25 by stephanosio