zephyrproject-rtos/sdk-ng #577: topic-picolibc: Cortex-R52 build doesn't accept several float instructions

keith-packard, 2022-10-18:
The Cortex-R52, which is an ARMv8-R architecture, appears to include a single-precision FPU, but the topic-picolibc toolchain doesn't compile this code:
```
float
rintf (float x)
{
  float result;
  __asm__ volatile ("vrintx.f32\t%0, %1" : "=t" (result) : "t" (x));
  return result;
}
```
it produces this error:
```
arm-zephyr-eabi-gcc -mcpu=cortex-r52 -mfloat-abi=hard -c foo.c
/tmp/ccADa46r.s: Assembler messages:
/tmp/ccADa46r.s:33: Error: unexpected character `w' in type specifier
/tmp/ccADa46r.s:33: Error: bad instruction `vrintx.f32.w s15,s15'
```
Tests with 0.15.1 work, tests with 0.15.1-12-g0599b1e fail. I can't see anything on that topic branch which could cause this failure.

keith-packard, 2022-10-18:
a second issue is that the toolchain doesn't appear to generate a multi-lib configuration which is compatible with this processor, ending up using `thumb/v7+fp/hard`

stephanosio, 2022-10-18:
> It looks like `-mfpu=vfp3-d16` makes the assembler fail to accept that instruction, which (I think) should be valid.

[According to GCC, ARMv8-R (Cortex-R52) has either `fpv5-sp-d16` (single precision configuration) or `neon-fp-armv8` (double precision + Advanced SIMD configuration)](https://github.com/gcc-mirror/gcc/blob/16e2427f50c208dfe07d07f18009969502c25dc8/gcc/config/arm/arm-cpus.in#L710-L724), so `vfpv3-d16` FPU type is incorrect.

I will send a Zephyr-side patch to specify the correct FPU type for the Cortex-R52.

> a second issue is that the toolchain doesn't appear to generate a multi-lib configuration which is compatible with this processor, ending up using `thumb/v7+fp/hard`

In fact, we (or the upstream GCC, for that matter) currently do not have any dedicated multilibs for non-M profile ARMv8 targets. This is mainly because ARMv8-A and ARMv8-R can execute ARMv7-A and ARMv7-R code as-is, and there is little to gain by having dedicated ARMv8 multilibs.

stephanosio, 2022-10-18:
Zephyr-side fix: https://github.com/zephyrproject-rtos/zephyr/pull/51377

closed 2022-10-18 by stephanosio

keith-packard, 2022-10-18:
Verified that this change enables these instructions in picolibc when built as a module