zephyrproject-rtos/sdk-ng #593: Add `long double` math function support in libc for all supported targets

stephanosio, 2022-11-15:
## Overview

Newlib (and picolibc) currently does not provide the `long double` math functions (i.e. the math functions suffixed with `l` such as `acoshl`) for `aarch64-zephyr-elf`, `riscv64-zephyr-elf` and `x86_64-zephyr-elf` targets because:

* Newlib does not implement any generic `long double` math functions.
* Newlib maps the `long double` math functions to the `double` math functions if and only if the type of `long double` is equivalent to the type of `double`.
    * [`long double` type check](https://github.com/zephyrproject-rtos/newlib-cygwin/blob/4e150303bcc1e44f4d90f3489a4417433980d5ff/newlib/configure.in#L673)
    * [`long double` math function mapping](https://github.com/zephyrproject-rtos/newlib-cygwin/blob/4e150303bcc1e44f4d90f3489a4417433980d5ff/newlib/libc/include/math.h#L414-L419)

On these targets, `typeof(long double) != typeof(double)`, so the `long double` math functions are not mapped to the `double` math functions, and none of the them provides target-specific `long double` math functions; leaving the `long double` math functions unimplemented.

For more details, refer to the issue #566.

## Implementation Strategy

Since our focus has shifted from the newlib to the picolibc, it would make the most sense to put the effort into getting the `long double` math function feature parity on the picolibc first.

Consider adding generic 128-bit (and possibly 80-bit) `long double` math function implementations from compatibly licensed open source libraries to the picolibc -- for example, [openlibm](https://github.com/JuliaMath/openlibm).

bojczmac, 2022-11-16:
I fixed above issue building my own SDK with this patch: https://github.com/espressif/newlib-esp32/commit/07be17637e8e3a796926c955d8361f3ddd8550a6

cfriedt, 2023-01-15:
@stephanosio - is this related?

keith-packard, 2023-01-15:
It looks like glibc (and hence libgcc) doesn't actually have soft float support for 80-bit values. Looks like we either need to fix that, or bail on long double support for -msoft-float 32-bit x86 targets.