zephyrproject-rtos/sdk-ng #616: internal compiler error: in is_base_type, at dwarf2out.cc:13400

heinwessels, 2023-01-05:
**Describe the bug**
We've run into a compiler crash while attempting to build zephyr with our C++ application. I have recreated this crash with a minimal example.
```
/home/hein/zephyrproject/zephyr/samples/subsys/cpp/cpp_synchronization/src/main.cpp: In substitution of 'template<class LambdaCaller>  requires  hasLambdaCaller<LambdaCaller> class Foo [with LambdaCaller = SomeLambdaCaller<>]':
/home/hein/zephyrproject/zephyr/samples/subsys/cpp/cpp_synchronization/src/main.cpp:27:23:   required from here
/home/hein/zephyrproject/zephyr/samples/subsys/cpp/cpp_synchronization/src/main.cpp:9:23: internal compiler error: in is_base_type, at dwarf2out.cc:13400
    9 |         {c.callLambda([]<typename TypeTag>(const TypeTag&) {})};
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
0x7fcf375fed8f __libc_start_call_main
        ../sysdeps/nptl/libc_start_call_main.h:58
0x7fcf375fee3f __libc_start_main_impl
        ../csu/libc-start.c:392
```

**Some notes**
- This crash popped up during the migration from Zephyr 2.6 (SDK 0.14.2) to Zephyr 3.2 (0.15.2). Using SDK 0.14.2 never caused an issue crash this.
- When building our application the error looks slightly different and registers as an `Segmentation fault`
```
internal compiler error: Segmentation fault
   12 |         {c.resolveType(id, []<typename TypeTag>(const TypeTag&) {})} -> std::same_as<Error>;
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
0x7f368628951f ???
        ./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0
0x7f3686270d8f __libc_start_call_main
        ../sysdeps/nptl/libc_start_call_main.h:58
0x7f3686270e3f __libc_start_main_impl
        ../csu/libc-start.c:392
```
**The minimal code**
```C++
#include <concepts>

#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(crash);

template <class C>
concept hasLambdaCaller = requires(C c) {
	/* Crashes on the following line */
	{c.callLambda([]<typename TypeTag>(const TypeTag&) {})};
}; 

template <typename SupportedComponents = void>
class SomeLambdaCaller {
public:
	template <typename F>
	void callLambda(F&& fn) const noexcept { }
};

template <hasLambdaCaller LambdaCaller>
class Foo {
public:
	/* Doesn't crash if this argument is removed */
	int foo(LambdaCaller& cl) noexcept { }
};

int main () {
	Foo<SomeLambdaCaller<>> Foo;
	SomeLambdaCaller caller;
    	Foo.foo(caller);
}
```
With following kconfig values
```kconfig
CONFIG_CPLUSPLUS=y
CONFIG_CPP_MAIN=y
CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=128
CONFIG_STD_CPP20=y
CONFIG_LIB_CPLUSPLUS=y
```

**Steps to recreate**
1. The minimal code construct is available on our fork, so clone https://github.com/fancom/zephyr/tree/attempt-recreate-crash
2. (Or simply recreate the application)
3. `west build -p always -b nucleo_h743zi samples/subsys/cpp/cpp_synchronization`
4. Observe crash

**Environment**
 - OS: ubuntu 22.04
 - Toolchain: zephyr SDK 0.15.2
 - Commit: https://github.com/zephyrproject-rtos/zephyr/commit/3210541c8658fd34068c254f81ce4f23ba0dc839

JvanDooren, 2023-01-05:
Note that the same code in GodBolt, compiled against GCC 12.1 causes the same internal compiler error! Changing it to GCC 12.2 fixes it, also stated in the release note for 12.2: https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

stephanosio, 2023-01-05:
Thanks for investigating the issue.

> Looks like we're in desperate need of a GCC version bump.

I will bump the GCC version to 12.2 in the upcoming 0.16.0 release.

PR #617 (2023-03-07): gcc: Pull in initial GCC 12.2 branch
This commit updates the gcc submodule to pull in the initial Zephyr GCC 12.2 branch with the Zephyr-specific patches (`zephyr-gcc-12.2` branch).

GCC PR: https://github.com/zephyrproject-rtos/gcc/pull/19

Fixes #616
closed 2023-03-07 by stephanosio