zephyrproject-rtos/sdk-ng #629: qemu_x86 interrupt enable delay causes test failure

galak, 2023-02-21:
qemu seems to take an extra instruction after an `sti` to post interrupts if there are one's pending.  This causes an issue when building the `tests/kernel/interrupt/arch.interrupt` for qemu_x86 w/LLVM.  Since LLVM code generation ends up having the instruction after the `sti` look at state that was suppose to be updated in the ISR.

Need to see why qemu doesn't post the interrupts after the 'sti' but before any other instructions execute.

galak, 2023-02-21:
Adding reference to https://github.com/zephyrproject-rtos/zephyr/pull/54689 that works around this issue in zephyr code base.

keith-packard, 2023-02-21:
STI has a defined 1-instruction interrupt shadow, is this more than that?

keith-packard, 2023-02-21:
it's pretty clear in docs I found about the sti instruction, https://cdrdv2-public.intel.com/671110/325383-sdm-vol-2abcd.pdf

galak, 2023-02-21:
@andyross @dcpleung comments on what @keith-packard is mentioning here about `sti` and 1-instruction delay.

dcpleung, 2023-02-21:
This is probably the case with the failed test. The fix would be to add a `nop` to `arch_irq_unlock()`.

andyross, 2023-02-21:
Huh, learn something new every day.  I don't think I ever knew this.  FWIW, I checked, and POPFL seems not to have this language.  Wonder if that's a mistake, or maybe on the early CPUs it was microcoded in such a way that the instruction queue would be drained anyway?

Does hardware actually exploit this?  Pipelines are a ton longer than 1 instruction on modern CPUs, they'd actually need extra hardware to implement "run one more instruction" than just to mark the STI as a stall.   I bet that you can't make this happen on hardware newer than a 286 or whatever.  Likewise... how does qemu get any benefit here?  Or maybe it's implementing the pedantic delay just to catch amateurs like us...

Definitely seems like we need to audit and make sure that every STI is followed by a NOP (or I guess some other instruction that is guaranteed not to depend on synchronized state, like a load of a constant or an unconditional jump/call, etc....)

andyross, 2023-02-21:
Amusingly apropos, this just went up today: https://www.righto.com/2023/02/8086-interrupt.html  (You guys are all reading this series, right?)

Bet you we could find the origin story in there.

galak, 2023-02-21:
Closing this issue since qemu is doing the right thing, and opened:

https://github.com/zephyrproject-rtos/zephyr/issues/55053

to track us fixing zephyr code base.

closed 2023-02-21 by galak

keith-packard, 2023-02-22:
> Huh, learn something new every day. I don't think I ever knew this. FWIW, I checked, and POPFL seems not to have this language. Wonder if that's a mistake, or maybe on the early CPUs it was microcoded in such a way that the instruction queue would be drained anyway?

My Intel architect friend says that there are only three instructions that have this 'interrupt shadow' behavior -- `sti`, `pop ss` and `mov ss`.
> Does hardware actually exploit this? Pipelines are a ton longer than 1 instruction on modern CPUs, they'd actually need extra hardware to implement "run one more instruction" than just to mark the [...]

andyross, 2023-02-22:
Interestingly the behavior with writes to SS came up recently in that Ken Shirrff blog too, so I knew that one.  There, they were to address what amounts to a mild state bug: when you change the stack segment but still have your old SP, your stack is instantaneously wrong (unless you use great care), so a processor interrupt that arrives at that moment will have its state pushed onto arbitrary memory.  The solution was to have hardware magically mask interrupt delivery for an instruction to make room to set the two registers atomically.

It's not clear why STI would need the same treatment, but maybe it was easier to have the two instructions use the same microcode states for whatever reason.

keith-packard, 2023-02-22:
> It's not clear why STI would need the same treatment, but maybe it was easier to have the two instructions use the same microcode states for whatever reason.

I was told it's very intentional (or retcon'd) for apps that want to use `popf; mangle-flags; sti; retf` instead of `iret`. Certainly these days x86 manufacturers don't seem to fear complexity...