zephyrproject-rtos/sdk-ng #631: ARC debug support broken after Zephyr SDK 0.14.2

jkiepert, 2023-03-01:
**Overview**
There appears to be a regression in ARC debugger support with Zephyr SDK releases after 0.14.2. Initially, we encountered this with our own ARC EM7D-based SoC, but I have also reproduced this regression with the natively supported Synopsys ARC EM Starter Kit (EMSK) board across Zephyr release versions when using Zephyr SDK 0.15.2 or later.

Board target: em_starterkit_em7d
Zephyr versions tested: v2.4, v2.7, v3.3
Zephyr SDK versions tested: 0.12.1, 0.14.2, 0.15.2, 0.16.0-beta1
Application tested: `zephyr/samples/basic/threads`

**Bug**
There appear to be two major regressions after SDK 0.14.2:
1. Debugger connection failure if `CONFIG_DEBUG_THREAD_INFO=y` (or `CONFIG_OPENOCD_SUPPORT=y` in earlier versions of Zephyr)
2. Code loading via GDB and code stepping via GDB is broken when connection is successful (only possible without `CONFIG_DEBUG_THREAD_INFO=y`)

It should be noted that code loading via `west flash` is functional with recent SDK versions. It is only debug functionality that is  broken (whether using west debug or cmake --target debug).

**To Reproduce**
Steps to reproduce the behavior debugger connection failure using Zephyr v3.3 (or v2.7) with SDK 0.16.0-beta1 or 0.15.2:
1. cd zephyr/samples/basic threads
3. west build -b em_starterkit_em7d -p always -- -DCMAKE_BUILD_TYPE=Debug -DCONFIG_DEBUG_THREAD_INFO=y
4. west debug
5. Output
```
(zephyr-v3.3) jkiepert@aurora:~/work/sdk/zephyr/samples/basic/threads$ west debug
-- west debug: rebuilding
ninja: no work to do.
-- west debug: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; thread info enabled
Open On-Chip Debugger 0.11.0+dev-00725-gc5c47943d (2022-12-07-08:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
Info : Zephyr: looking for target: arcv2
openocd: src/jtag/core.c:338: jtag_checks: Assertion `jtag_trst == 0' failed.
GNU gdb (Zephyr SDK 0.16.0-beta1) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build/zephyr/zephyr.elf...
:3333: Connection timed out.
You can't do that when your target is `exec'
(gdb) 
```
Notice the JTAG failure message:
```
openocd: src/jtag/core.c:338: jtag_checks: Assertion `jtag_trst == 0' failed.
...
:3333: Connection timed out.
You can't do that when your target is `exec'
```
Steps to reproduce the behavior GDB load/step failure using Zephyr v3.3 (or v2.7) + SDK 0.16.0-beta1 or 0.15.2
1. cd zephyr/samples/basic threads
3. west build -b em_starterkit_em7d -p always -- -DCMAKE_BUILD_TYPE=Debug
4. west debug
5. Output
```
(zephyr-v3.3) jkiepert@aurora:~/work/sdk/zephyr/samples/basic/threads$ west debug
-- west debug: rebuilding
ninja: no work to do.
-- west debug: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
Open On-Chip Debugger 0.11.0+dev-00725-gc5c47943d (2022-12-07-08:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
Info : clock speed 5000 kHz
Info : JTAG tap: arc-em.cpu tap/device found: 0x200044b1 (mfg: 0x258 (ARC International), part: 0x0004, ver: 0x2)
Info : starting gdb server for arc-em.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* arc-em.cpu         arcv2      little arc-em.cpu         halted

Info : Listening on port 6333 for tcl connections
Info : Listening on port 4444 for telnet connections
GNU gdb (Zephyr SDK 0.16.0-beta1) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build/zephyr/zephyr.elf...
Remote debugging using :3333
Info : accepting 'gdb' connection on tcp/3333
0x000006bc in sys_heap_aligned_alloc (heap=<optimized out>, align=0, bytes=0) at /home/jkiepert/work/sdk/zephyr/lib/os/heap.c:344
/usr/lib/python3/dist-packages/traitlets/config/loader.py:795: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if len(key) is 1:
/usr/lib/python3/dist-packages/traitlets/config/loader.py:804: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if len(key) is 1:
344             mem = (uint8_t *) ROUND_UP(mem + rew, align) - rew;
Warn : No RTOS could be auto-detected!
Loading section text, size 0x5894 lma 0x0
Load failed
(gdb) load
Loading section text, size 0x5894 lma 0x0
Load failed
(gdb) list
349             chunkid_t c_end = end - chunk_buf(h);
350             CHECK(c >= c0 && c  < c_end && c_end <= c0 + padded_sz);
351
352             /* Split and free unused prefix */
353             if (c > c0) {
354                     split_chunks(h, c0, c);
355                     free_list_add(h, c0);
356             }
357
358             /* Split and free unused suffix */
(gdb) s
^CInfo : The target is not running when halt was requested, stopping GDB.
Info : The target is not running when halt was requested, stopping GDB.
```
At this point openocd continues to spew that last line until openocd is manually killed (`killall -9 openocd`).

**Expected behavior**
Connecting the debugger with `west debug` should halt the CPU and allow one to load code, single step the processor, and when CONFIG_DEBUG_THREAD_INFO=y, openocd should detect Zephyr and provide thread awareness such as `info threads`. All of this functionality is present when using Zephyr SDK 0.14.2 and 0.12.1.

**Impact**
This is a showstopper for Zephyr ARC development with Zephyr SDKs after 0.14.2.

**Logs and console output**
See above

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.0-beta1 and 0.15.2
 - Commit SHA or Version used: Zephyr tag v3.3.0 (07c6af3b8c35c1e49186578ca61a25c76e2fb308)

**Additional context**
The EMSK board is configured for the EM7D CPU with the DIP switch set to OFF,OFF,OFF,OFF. However, the behavior appears to be the same independent of the particular ARC core used.

abrodkin, 2023-03-01:
@jkiepert thanks for the report, let us look at what's wrong here. I'm wondering if this issue is blocking any important activities on your side, i.e. how soon that needs to be resolved?

jkiepert, 2023-03-01:
@abrodkin thank you for taking a look into this issue. To answer your question, this issue mainly prevents us from migrating to Zephyr v3.3 for our ARC-based products since Zephyr v3.3 requires SDK 0.15+. For the time being, we will stay on SDK 0.14.2 and Zephyr v2.7 until things have been fixed in the later Zephyr SDK versions.

jkiepert, 2023-03-02:
@abrodkin I did a bit further investigation and found that there is a Zephyr version dependency to this issue as well. Specifically, I moved to the minimum SDK supported by Zephyr v2.7 (SDK 0.13.1) and then tested `west debug` with `CONFIG_OPENOCD_SUPPORT=y` across zephyr versions between v2.4 and v2.7 with the `em_starterkit_em7d` board target.

I found that the debugger connection and openocd thread awareness works for all zephyr versions between v2.4 and v2.6, but it breaks with Zephyr v2.7 and above. I do see a crash report generated for openocd. I've attached an example crash with 0.13.1, but similar crashes occur for SDK 0.14.2, 0.15.2, and 0.16.0-beta1.

Here's an example snippet from the 0.13.1 openocd crash report (report is similar for newer SDK versions):
```
ProblemType: Crash
Architecture: amd64
CurrentDesktop: KDE
Date: Wed Mar  1 17:17:09 2023
DistroRelease: Ubuntu 18.04
ExecutablePath: /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
ExecutableTimestamp: 1677716162
ProcCmdline: /opt/tools/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/jkiepert/work/sdk/zephyr/boards/arc/em_starterkit/support -s /opt/tools/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/jkiepert/work/sdk/zephyr/boards/arc/em_starterkit/support/openocd.cfg -c tcl_port\ 6333 -c telnet_port\ 4444 -c gdb_port\ 3333 -c $_TARGETNAME\ configure\ -rtos\ Zephyr -c init -c targets -c\ halt
ProcCwd: /home/jkiepert/work/sdk/zephyr/samples/basic/threads
ProcEnviron:
 LANG=en_US.UTF-8
 TERM=xterm-256color
 SHELL=/bin/bash
 LANGUAGE=
 XDG_RUNTIME_DIR=<set>
 PATH=(custom, user)
ProcMaps:
 55fd4fbde000-55fd4fc3a000 r--p 00000000 103:05 10888738                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
 55fd4fc3a000-55fd4fe36000 r-xp 0005c000 103:05 10888738                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
 55fd4fe36000-55fd4ff35000 r--p 00258000 103:05 10888738                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
 55fd4ff35000-55fd4ff71000 r--p 00356000 103:05 10888738                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
 55fd4ff71000-55fd4ff88000 rw-p 00392000 103:05 10888738                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/openocd
 55fd4ff88000-55fd50009000 rw-p 00000000 00:00 0 
 55fd510b9000-55fd51160000 rw-p 00000000 00:00 0                          [heap]
 7f366ff4c000-7f366ff4e000 rw-p 00000000 00:00 0 
 7f366ff4e000-7f366ff55000 r--p 00000000 103:05 10888657                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libpthread-2.32.so
 7f366ff55000-7f366ff64000 r-xp 00007000 103:05 10888657                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libpthread-2.32.so
 7f366ff64000-7f366ff6a000 r--p 00016000 103:05 10888657                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libpthread-2.32.so
 7f366ff6a000-7f366ff6b000 r--p 0001b000 103:05 10888657                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libpthread-2.32.so
 7f366ff6b000-7f366ff6c000 rw-p 0001c000 103:05 10888657                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libpthread-2.32.so
 7f366ff6c000-7f366ff70000 rw-p 00000000 00:00 0 
 7f366ff70000-7f366ff75000 r--p 00000000 103:05 10888666                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libudev.so.1.6.3
 7f366ff75000-7f366ff8a000 r-xp 00005000 103:05 10888666                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libudev.so.1.6.3
 7f366ff8a000-7f366ff93000 r--p 0001a000 103:05 10888666                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libudev.so.1.6.3
 7f366ff93000-7f366ff94000 ---p 00023000 103:05 10888666                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libudev.so.1.6.3
 7f366ff94000-7f366ff95000 r--p 00023000 103:05 10888666                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libudev.so.1.6.3
 7f366ff95000-7f366ff96000 rw-p 00024000 103:05 10888666                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libudev.so.1.6.3
 7f366ff96000-7f366ffbc000 r--p 00000000 103:05 10888641                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libc-2.32.so
 7f366ffbc000-7f36700fe000 r-xp 00026000 103:05 10888641                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libc-2.32.so
 7f36700fe000-7f367014b000 r--p 00168000 103:05 10888641                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libc-2.32.so
 7f367014b000-7f367014c000 ---p 001b5000 103:05 10888641                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libc-2.32.so
 7f367014c000-7f367014f000 r--p 001b5000 103:05 10888641                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libc-2.32.so
 7f367014f000-7f3670155000 rw-p 001b8000 103:05 10888641                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libc-2.32.so
 7f3670155000-7f367015b000 rw-p 00000000 00:00 0 
 7f367015b000-7f367015c000 r--p 00000000 103:05 10888643                  /opt/tools/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/lib/libdl-2.32.so
[... cut ...]
 Mems_allowed:	00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000001
 Mems_allowed_list:	0
 voluntary_ctxt_switches:	2
 nonvoluntary_ctxt_switches:	0
Signal: 6
Uname: Linux 4.15.0-200-generic x86_64
UserGroups: dialout disk docker lpadmin sudo users vboxusers
_LogindSession: 3
```

kokas-a, 2023-03-07:
Hi @jkiepert,

We are still working on the issue, but I'd like to suggest you workaround.

If you type ```west -v debug``` you may see that west creates 2 separate processes:
openocd and gdb with appropriate sets of parameters.
One of OpenOcd's parameter is a command ``` -c '$_TARGETNAME configure -rtos Zephyr'``` and
it is located in the middle of parameters list. 
This parameters order cause error: ```openocd: src/jtag/core.c:338: jtag_checks: Assertion `jtag_trst == 0' failed.```

Suggestion is to make a manual call of openocd and move the os-awareness command at the and of the argument list. 
It may looks like:
```
<PATH_TO_SDK>/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/openocd 
-s <PATH_TO_ZEPHYR>/zephyrproject/zephyr/boards/arc/em_starterkit/support 
-s <PATH_TO_SDK>/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts  
-f <PATH_TO_ZEPHYR>/zephyrproject/zephyr/boards/arc/em_starterkit/support/openocd.cfg 
-c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333' '-c init' '-c targets' '-c halt' 
-c '$_TARGETNAME configure -rtos Zephyr'
```

In parallel GDB should be executed with unchanged list of parameters:
```
<PATH_TO_SDK>/zephyr-sdk-0.15.0/arc-zephyr-elf/bin/arc-zephyr-elf-gdb 
-ex 'target extended-remote :3333' 
<PATH_TO_ZEPHYR>/zephyrproject/zephyr/samples/basic/threads/build/zephyr/zephyr.elf -ex load
```

jkiepert, 2023-03-07:
Thanks @kokas-a, I'll give this a try.

kokas-a, 2023-03-16:
> 1. Debugging without thread awareness fails to load code at all:

Could you please provide output of ```west -v debug``` command (with verbose key)? It's not clear from your log what kind of parameters were passed to OpenOCD and GDB.

perceiver-josh, 2023-03-16:
@kokas-a Here's the log with `-v`:
```
$ west -v debug
-- west debug: rebuilding
cmake version 3.20.5 is OK; minimum version is 3.13.1
Running CMake: /usr/local/bin/cmake --build /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build
ninja: no work to do.
-- west debug: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
runners.openocd: /opt/tools/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/jkiepert/work/sdk/zephyr/boards/arc/em_starterkit/support -s /opt/tools/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/jkiepert/work/sdk/zephyr/boards/arc/em_starterkit/support/openocd.cfg -c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333' '-c init' '-c targets' '-c halt'
runners.openocd: /opt/tools/zephyr-sdk/arc-zephyr-elf/bin/arc-zephyr-elf-gdb-py -ex 'target extended-remote :3333' /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build/zephyr/zephyr.elf -ex load
Open On-Chip Debugger 0.11.0+dev-00725-gc5c47943d (2022-11-22-06:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
Info : clock speed 5000 kHz
Info : JTAG tap: arc-em.cpu tap/device found: 0x200044b1 (mfg: 0x258 (ARC International), part: 0x0004, ver: 0x2)
Info : starting gdb server for arc-em.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* arc-em.cpu         arcv2      little arc-em.cpu         halted

Info : Listening on port 6333 for tcl connections
Info : Listening on port 4444 for telnet connections
GNU gdb (Zephyr SDK 0.15.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build/zephyr/zephyr.elf...
Remote debugging using :3333
Info : accepting 'gdb' connection on tcp/3333
0x0000cf14 in k_sched_unlock () at /home/jkiepert/work/sdk/zephyr/kernel/sched.c:1003
1003			__ASSERT(_current->base.sched_locked != 0U, "");
Warn : No RTOS could be auto-detected!
Loading section text, size 0xf8f8 lma 0x0
Load failed
(gdb) 
```

perceiver-josh, 2023-03-16:
@kokas-a interestingly, if I manually run openocd and gdb I see non-thread-aware debug loading work. The key is to not pass in `-ex load` with the gdb command.

Proceedure that seems to work for `CONFIG_DEBUG_THEAD_INFO=n`:
```
/opt/tools/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/jkiepert/work/sdk/zephyr/boards/arc/em_starterkit/support -s /opt/tools/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/jkiepert/work/sdk/zephyr/boards/arc/em_starterkit/support/openocd.cfg -c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333' '-c init' '-c targets' '-c halt'
```
In another terminal:
```
$ /opt/tools/zephyr-sdk/arc-zephyr-elf/bin/arc-zephyr-elf-gdb-py -ex 'target extended-remote :3333' /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build/zephyr/zephyr.elf
GNU gdb (Zephyr SDK 0.15.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build/zephyr/zephyr.elf...
Remote debugging using :3333
0x0000cf14 in k_sched_unlock () at /home/jkiepert/work/sdk/zephyr/kernel/sched.c:1003
1003			__ASSERT(_current->base.sched_locked != 0U, "");
(gdb) load
Loading section text, size 0xf8f8 lma 0x0
Loading section initlevel, size 0x88 lma 0xf8f8
Loading section devices, size 0xc0 lma 0xf980
Loading section sw_isr_table, size 0xb0 lma 0xfa40
Loading section _static_thread_data_area, size 0x90 lma 0xfaf0
Loading section device_handles, size 0x30 lma 0xfb80
Loading section log_const_sections, size 0x20 lma 0xfbb0
Loading section rodata, size 0x1d90 lma 0xfbd0
Loading section datas, size 0xe4 lma 0x80001c88
Loading section device_states, size 0x10 lma 0x80001d6c
Loading section k_heap_area, size 0x18 lma 0x80001d7c
Loading section k_queue_area, size 0x14 lma 0x80001d94
Start address 0x000040cc, load size 72320
Transfer rate: 446 KB/sec, 4821 bytes/write.
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
arch_cpu_idle () at /home/jkiepert/work/sdk/zephyr/arch/arc/core/cpu_idle.S:49
49		j_s [blink]
(gdb) s
k_cpu_idle () at ../../../../include/zephyr/kernel.h:5692
5692	}
(gdb) 
idle (unused1=0x8000014c <_kernel>, unused2=0x0 <_VectorTable>, unused3=0x0 <_VectorTable>) at /home/jkiepert/work/sdk/zephyr/kernel/idle.c:50
50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) n
62			(void) arch_irq_lock();
(gdb) 
86			k_cpu_idle();
(gdb) 
50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) 
62			(void) arch_irq_lock();
(gdb) 
86			k_cpu_idle();
(gdb) 
50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) 
62			(void) arch_irq_lock();
(gdb) 
86			k_cpu_idle();
(gdb) 
50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) 
62			(void) arch_irq_lock();
(gdb) 
86			k_cpu_idle();
(gdb) 

50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) 
62			(void) arch_irq_lock();
(gdb) 
86			k_cpu_idle();
(gdb) 

50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) 
62			(void) arch_irq_lock();
(gdb) 
86			k_cpu_idle();
(gdb) 
50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) 
62			(void) arch_irq_lock();
(gdb) 
86			k_cpu_idle();
(gdb) 
50			if (IS_ENABLED(CONFIG_SMP) && !IS_ENABLED(CONFIG_SCHED_IPI_SUPPORTED)) {
(gdb) 
62			(void) arch_irq_lock();
(gdb) quit
A debugging session is active.

	Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) y
Detaching from program: /home/jkiepert/work/sdk/zephyr/samples/basic/threads/build/zephyr/zephyr.elf, Remote target
[Inferior 1 (Remote target) detached]
```