zephyrproject-rtos/sdk-ng #639: Picolibc build fails for microblazeel

stephanosio, 2023-03-08:
Picolibc build fails for the `microblazeel-zephyr-elf` target:

```
[INFO ]  Installing Picolibc library
[DEBUG]    Entering '/__w/_temp/workspace/build/.build/microblazeel-zephyr-elf/build/build-picolibc-build-x86_64-build_pc-linux-gnu'
[EXTRA]    Configuring Picolibc library
[DEBUG]    ==> Executing:  'meson' '--cross-file' 'picolibc-cross.txt' '--prefix=/__w/_temp/workspace/output/microblazeel-zephyr-elf' '-Dincludedir=picolibc/include' '-Dlibdir=picolibc/microblazeel-zephyr-elf/lib' '-Dspecsdir=/__w/_temp/workspace/output/microblazeel-zephyr-elf/microblazeel-zephyr-elf/lib' '/__w/_temp/workspace/build/.build/microblazeel-zephyr-elf/src/picolibc' '-Dio-c99-formats=true' '-Dio-long-long=false' '-Dnewlib-register-fini=false' '-Dnewlib-nano-malloc=true' '-Dnewlib-atexit-dynamic-alloc=false' '-Dnewlib-global-atexit=true' '-Dlite-exit=true' '-Dnewlib-multithread=true' '-Dnewlib-retargetable-locking=true' '-Dthread-local-storage=auto' '-Derrno-function=zephyr' '-Dsysroot-install=true' '-Dsysroot-install-skip-checks=true'
[CFG  ]    WARNING: Unknown CPU family microblazeel, please report this at https://github.com/mesonbuild/meson/issues/new
[CFG  ]    DEPRECATION: c_args in the [properties] section of the machine file is deprecated, use the [built-in options] section.
[CFG  ]    The Meson build system
[CFG  ]    Version: 0.61.5
[CFG  ]    Source dir: /__w/_temp/workspace/build/.build/microblazeel-zephyr-elf/src/picolibc
[CFG  ]    Build dir: /__w/_temp/workspace/build/.build/microblazeel-zephyr-elf/build/build-picolibc-build-x86_64-build_pc-linux-gnu
[CFG  ]    Build type: cross build
[CFG  ]    Project name: picolibc
[CFG  ]    Project version: 1.8
[CFG  ]    Cross compiler sanity tests disabled via the cross file.
[CFG  ]    C compiler for the host machine: microblazeel-zephyr-elf-gcc (gcc 12.2.0 "microblazeel-zephyr-elf-gcc (Zephyr SDK 0.93.0-120-g444a430) 12.2.0")
[CFG  ]    C linker for the host machine: microblazeel-zephyr-elf-gcc ld.bfd 0.93.0-120
[CFG  ]    C compiler for the build machine: cc (gcc 7.5.0 "cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
[CFG  ]    C linker for the build machine: cc ld.bfd 2.30
[CFG  ]    Build machine cpu family: x86_64
[CFG  ]    Build machine cpu: x86_64
[CFG  ]    Host machine cpu family: microblazeel
[CFG  ]    Host machine cpu: microblazeel
[CFG  ]    Target machine cpu family: microblazeel
[CFG  ]    Target machine cpu: microblazeel
[CFG  ]    Compiler for C supports arguments -nostdlib: YES 
[CFG  ]    
[CFG  ]    ../../src/picolibc/meson.build:112:2: ERROR: Problem encountered: 
[CFG  ]    
[CFG  ]    Unsupported architecture: "microblazeel"
[CFG  ]    
[CFG  ]        Read the Supported Architectures section in README.md
[CFG  ]        to learn how to add a new architecture.
[CFG  ]    
[CFG  ]    
[CFG  ]    A full log can be found at /__w/_temp/workspace/build/.build/microblazeel-zephyr-elf/build/build-picolibc-build-x86_64-build_pc-linux-gnu/meson-logs/meson-log.txt
[CFG  ]    NOTICE: You are using Python 3.6 which is EOL. Starting with v0.62.0, Meson will require Python 3.7 or newer
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing Picolibc library'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@376]
[ERROR]  >>        called from: do_picolibc_for_target[scripts/build/companion_libs/340-picolibc.sh@149]
[ERROR]  >>        called from: do_companion_libs_for_target[scripts/build/companion_libs.sh@43]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@697]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      https://crosstool-ng.github.io/docs/known-issues/
[ERROR]  >>
[ERROR]  >> NOTE: Your configuration includes features marked EXPERIMENTAL.
[ERROR]  >> Before submitting a bug report, try to reproduce it without enabling
[ERROR]  >> any experimental features. Otherwise, you'll need to debug it
[ERROR]  >> and present an explanation why it is a bug in crosstool-NG - or
[ERROR]  >> preferably, a fix.
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]  
```

Full build log: [log_linux-x86_64_microblazeel-zephyr-elf.zip](https://github.com/zephyrproject-rtos/sdk-ng/files/10921261/log_linux-x86_64_microblazeel-zephyr-elf.zip)

Picolibc will need to be supported before the Zephyr SDK microblazeel toolchain can be considered stable and the microblaze architecture support can be merged.

stephanosio, 2023-03-08:
cc @keith-packard

alpsayin, 2023-03-08:
> Picolibc will need to be supported before the Zephyr SDK microblazeel toolchain can be considered stable and the microblaze architecture support can be merged.

I'm sorry. I couldn't fully understand this statement. Are you saying, picolibc should merge before microblaze support is merged?
If yes, I agree. Making microblaze support picolibc (if any additional work becomes necessary) should be part of the microblaze work and should not block the greater picture here.

Thus, the next step here would be to blacklist microblazeel from picolibc build process, right?

stephanosio, 2023-03-08:
@alpsayin Picolibc support is already part of the upstream Zephyr and it will soon become the default libc for Zephyr; meaning, if you want to upstream Microblaze support, it will need to have Picolibc support (i.e. Picolibc will need to be able to build for it).

stephanosio, 2023-03-08:
> Thus, the next step here would be to blacklist microblazeel from picolibc build process, right?

Yes, that is done here: https://github.com/stephanosio/zephyr-sdk-ng/commit/88a8937e8130a05b358ca82d32365e4ece671eb0 (note that the microblazeel config file did not enable Picolibc, when it should have).

alpsayin, 2023-03-08:
Ah okay, got it, much clearer now. Thank you very much.
The issue then seem to be simply resolvable by adding microblaze to support cpus list.
What comes next is a different story 🙃

alpsayin, 2023-03-08:
Okay, it's a bit more than that. I'm on this.

keith-packard, 2023-03-08:
> Ah okay, got it, much clearer now. Thank you very much. The issue then seem to be simply resolvable by adding microblaze to support cpus list. What comes next is a different story upside_down_face

The cmake code for picolibc (used while building picolibc as a module) allows "unsupported" architectures to build, but the meson code (as used by the SDK) checks to make sure someone has actually validated that the target is supported and has all of the bits necessary for complete C language support. That requires setjmp/longjmp code, but can also include thread local storage helpers, math exception definitions, qemu startup code and as much architecture-specific acceleration as you like. This is covered in the picolibc documentation here:

https://github.com/picolibc/picolibc#supported-architectures

alpsayin, 2023-03-08:
Droping links:

1. `picolibc` pull requested to upstream: https://github.com/zephyrproject-rtos/picolibc/pull/441
1. `sdk-ng` draft pull request pointing to above PR to test build: https://github.com/zephyrproject-rtos/sdk-ng/pull/641

keith-packard, 2025-04-04:
This was fixed in picolibc 1.8.2

closed 2025-05-03 by nashif