zephyrproject-rtos/sdk-ng #646: microblaze: atomic compare-and-swap returns incorrect value

alpsayin, 2023-03-13:
Microblaze toolchain's atomic compare-and-swap returns the inverse boolean value causing at least `ungetc` implementation in `picolibc` to hang.
The issue was fixed in Xilinx's gcc via the below patch but this has not made it to upstream and thus not to zephyr-sdk.
https://github.com/Xilinx/meta-xilinx/blob/master/meta-microblaze/recipes-devtools/gcc/gcc-12/0009-Patch-microblaze-Fix-atomic-boolean-return-value.patch
CC @keith-packard

keith-packard, 2023-03-13:
I'm still seeing problems with atomic_exchange use in getc -- it fails to return the previously stored value as it should. Picolibc now has a stand-alone atomics test case that can be used to verify that atomics work as needed for picolibc.

alpsayin, 2023-10-01:
Will be fixed with https://github.com/zephyrproject-rtos/gcc/pull/24

closed 2025-05-03 by nashif