zephyrproject-rtos/sdk-ng #669: internal compiler error: arm_mat_mult_f64

XenuIsWatching, 2023-04-26:
On https://github.com/zephyrproject-rtos/zephyr/pull/56957, an internal compiler is shown.
```
/__w/zephyr/modules/lib/cmsis-dsp/Source/MatrixFunctions/arm_mat_mult_f64.c: In function 'arm_mat_mult_f64':
/__w/zephyr/modules/lib/cmsis-dsp/Source/MatrixFunctions/arm_mat_mult_f64.c:506:1: internal compiler error: in arc_split_move, at config/arc/arc.cc:10235
  506 | }
      | ^
0x7ff86c016d8f __libc_start_call_main
	../sysdeps/nptl/libc_start_call_main.h:58
0x7ff86c016e3f __libc_start_main_impl
	../csu/libc-start.c:392
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://github.com/zephyrproject-rtos/sdk-ng/issues> for instructions.
```

This is with the CMSIS-DSP found here https://github.com/ARM-software/CMSIS-DSP/blob/main/Source/MatrixFunctions/arm_mat_mult_f64.c

The Command Line (which just comes straight from the CI log)
```
 /opt/toolchains/zephyr-sdk-0.16.0/arc-zephyr-elf/bin/arc-zephyr-elf-gcc -DARM_MATH_LOOPUNROLL -DDISABLEFLOAT16 -DKERNEL -DTC_RUNID=c973cdd3ec5ea82bd9756a24c4fc1eb9 -D_ANSI_SOURCE -D_FORTIFY_SOURCE=1 -D__GNUC_PYTHON__ -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/__w/zephyr/modules/lib/cmsis-dsp/Source/DistanceFunctions -I/__w/zephyr/modules/lib/cmsis-dsp/Source/../Include -I/__w/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/__w/zephyr/modules/lib/cmsis-dsp/Source/../PrivateInclude -I/__w/zephyr/zephyr/include -I/__w/zephyr/zephyr/twister-out/nsim_sem/tests/subsys/dsp/basicmath/zdsp.basicmath.fpu/zephyr/include/generated -I/__w/zephyr/zephyr/soc/arc/snps_nsim -I/__w/zephyr/zephyr/lib/libc/newlib/include -I/__w/zephyr/zephyr/subsys/testsuite/include -I/__w/zephyr/zephyr/subsys/testsuite/ztest/include -I/__w/zephyr/zephyr/subsys/dsp/cmsis/public -I/__w/zephyr/modules/lib/cmsis-dsp/Include -I/__w/zephyr/modules/lib/cmsis-dsp/PrivateInclude -fno-strict-aliasing -Werror -Os -imacros /__w/zephyr/zephyr/twister-out/nsim_sem/tests/subsys/dsp/basicmath/zdsp.basicmath.fpu/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=em4_fpuda --sysroot=/opt/toolchains/zephyr-sdk-0.16.0/arc-zephyr-elf/arc-zephyr-elf -imacros /__w/zephyr/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/__w/zephyr/zephyr/tests/subsys/dsp/basicmath=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -g3 -gdwarf-2 -fno-delete-null-pointer-checks -mtp-regno=none -mno-sdata -mmpy-option=wlh1 -mfpu=fpuda_all -std=c99 -Ofast -MD -MT modules/cmsis-dsp/CMakeFiles/..__modules__lib__cmsis-dsp.dir/Source/MatrixFunctions/arm_mat_mult_f64.c.obj -MF modules/cmsis-dsp/CMakeFiles/..__modules__lib__cmsis-dsp.dir/Source/MatrixFunctions/arm_mat_mult_f64.c.obj.d -o modules/cmsis-dsp/CMakeFiles/..__modules__lib__cmsis-dsp.dir/Source/MatrixFunctions/arm_mat_mult_f64.c.obj -c /__w/zephyr/modules/lib/cmsis-dsp/Source/MatrixFunctions/arm_mat_mult_f64.c
``

stephanosio, 2023-04-26:
cc @abrodkin @evgeniy-paltsev @ruuddw @claziss

stephanosio, 2023-05-02:
Setting low priority since https://github.com/zephyrproject-rtos/zephyr/pull/56957, where this bug was discovered, found a workaround to make this ICE not show up (compile without the unrolled loops).