zephyrproject-rtos/sdk-ng #675: libstdc++: Use of non-standard memalign instead of C11 aligned_alloc

stephanosio, 2023-05-18:
libstdc++ seems to internally make use of the GNU `memalign` function instead of the ISO C `aligned_alloc` function.

[Looking at the libstdc++ internals, it should be using the `aligned_alloc` instead of `memalign` when `aligned_alloc` is detected as available](https://github.com/gcc-mirror/gcc/blob/5d85b5d649fff675ff00adcc99371bccf4ef5944/libstdc%2B%2B-v3/libsupc%2B%2B/new_opa.cc#L66-L67):

```c++
namespace __gnu_cxx {
#if _GLIBCXX_HAVE_ALIGNED_ALLOC
using ::aligned_alloc;
[...]
#elif _GLIBCXX_HAVE_MEMALIGN
static inline void*
aligned_alloc (std::size_t al, std::size_t sz)
{
  // Solaris requires al >= sizeof a word and QNX requires >= sizeof(void*)
  // but they both provide posix_memalign, so will use the definition above.
  return memalign (al, sz);
}
[...]
  while ((p = __gnu_cxx::aligned_alloc (align, sz)) == nullptr)
```

Find out why `_GLIBCXX_HAVE_ALIGNED_ALLOC` is not being defined and make sure it is defined and the non-standard `memalign` function is not used by libstdc++.

keith-packard, 2023-05-18:
Good debugging -- it looks like gcc is just missing an `AC_DEFINE(HAVE_ALIGNED_ALLOC)` in `libstdc++-v3/configure.ac` in the newlib section. It has no mechanism to autodetect the available libc API, so it hard-codes the list of available functions and `aligned_alloc()` is just missing here. Super easy to fix; I'll get a patch made for the SDK.

However, I don't think that eliminates the need for the `memalign()` alias in the common C library -- every gcc-based toolchain which doesn't have the necessary patch will fail unless that alias is provided. If we can detect when a fixed toolchain is in use, we can skip the definition there.

keith-packard, 2023-05-18:
The proposed change is here: https://github.com/zephyrproject-rtos/gcc/pull/22
I haven't tested it.