zephyrproject-rtos/sdk-ng #691: Thread awareness is not available on QEMU targets

bdmendes, 2023-07-28:
**Describe the bug**

`gdb` is unable to show threads list while debugging an executable on Zephyr `QEMU` targets. Instead of full thread list only the one running thread (current thread) is shown.

This is only an issue with the various Zephyr `QEMU` targets; thread awareness works as expected with the hardware targets I have tested, through `OpenOCD`.

**To Reproduce**

Steps to reproduce the behavior:
1. west build -b qemu_x86 zephyr/samples/philosophers/
2. ninja debugserver
3. [zephyrsdk path]/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gdb zephyr/build/zephyr/zephyr.elf
4. Halt execution and issue `info threads`
5. See only one running thread

**Expected behavior**

A complete list of all program threads shown, preferably with their name and priority.

**Impact**

This makes it harder to debug highly concurrent applications for issues such as thread synchronization.

**Logs and console output**

Expected output (example from board `iotdk`):

```sh
west debug
(gdb) continue
(gdb) ^C
(gdb) info threads
  Id   Target Id                                                                   Frame
* 2    Thread 2147484656 "idle" (Name: idle, prio:40,useropts:1)                   arch_cpu_idle ()
    at /home/mendesb/zephyrproject/zephyr/arch/arc/core/cpu_idle.S:49
  3    Thread 2147484368 "Philosopher 5" (Name: Philosopher 5, prio:-2,useropts:4) 0x00003ce2 in ?? ()
  4    Thread 2147484224 "Philosopher 4" (Name: Philosopher 4, prio:-1,useropts:4) 0x00003ce2 in ?? ()
  5    Thread 2147484080 "Philosopher 3" (Name: Philosopher 3, prio:0,useropts:4)  0x00003ce2 in ?? ()
  6    Thread 2147483936 "Philosopher 2" (Name: Philosopher 2, prio:1,useropts:4)  0x00003ce2 in ?? ()
  7    Thread 2147483792 "Philosopher 1" (Name: Philosopher 1, prio:2,useropts:4)  0x00003ce2 in ?? ()
  8    Thread 2147483648 "Philosopher 0" (Name: Philosopher 0, prio:3,useropts:4)  0x00003ce2 in ?? ()
```

Actual output from `qemu_x86`:

```sh
(gdb) continue
(gdb) ^C
(gdb) info threads
  Id   Target Id                    Frame
* 1    Thread 1.1 (CPU#0 [running]) 0xe61db94c in ?? ()
```

**Environment**
 - OS: Ubuntu 22.04
 - Toolchain: Zephyr SDK
 - Zephyr Commit SHA: fe15f7736c

bdmendes, 2023-08-07:
After digging through `QEMU`'s `gdbserver` (`gdbstub.c`), it appears to me that the "threads" it exposes are a one-to-one relationship with the emulated cores, and not the target's OS threads. As such, a lot of additional infrastructure needs to be built into `QEMU` for OS/plataform-specific inspection.
https://github.com/zephyrproject-rtos/zephyr/discussions/57816 might be a relevant discussion.

lmapii, 2024-08-15:
This is a bit of a bugger right now for debugging integration tests related to threads on anything but Linux since the `native_posix` target is only supported on Linux. So right now on macos or Windows the only way to get thread awareness when testing is via an actual board (?)