zephyrproject-rtos/sdk-ng #695: arc soft float multiply error

keith-packard, 2023-08-14:
I found one case where the soft 32-bit float multiplication function, `__mulsf3`, for arc is generating an incorrect result.

    0x1.000002p-2f * 0x1.000002p-126f

The exact result is 0x1.000004000004p-128, or 0x0.400001000001p-126. This should get rounded to 0x0.400002p-126, but the arc soft float code returns 0x0.4p-126.

I've tested soft float on arm and riscv and they both work correctly, along with various hardware floating point devices as well

abrodkin, 2023-08-15:
Cc @claziss

claziss, 2023-08-15:
Please @keith-packard  can you tell me for which architecture you see this, and what gcc version u use?

keith-packard, 2023-08-15:
GCC for Zephyr SDK 0.16.2-rc1 (12.2.0). arcem, archs and em architectures. I'm pretty sure it's just a bug in the custom arc soft float assembly implementation.