zephyrproject-rtos/sdk-ng #709: picolibc: Soft-float math functions access the FPU on SPARC

tbr-tt, 2023-10-03:
In SDK 0.16.3, it appears that the fenv.h functions of the picolibc soft-float build for SPARC access the floating point status register (`%fsr`). One example is `fegetround()`. That is not compatible running on a processor without FPU or with FPU disabled.

One way to observe the issue is to scan for accesses to the `%fsr` register in the object disassembly.

There are some FPU accesses in Picolib fenv functions (unexpected):
```
cd /opt/zephyr-sdk-0.16.3/sparc-zephyr-elf/
./bin/sparc-zephyr-elf-objdump -d picolibc/sparc-zephyr-elf/lib/leon/soft/libc.a | grep %fsr | wc -l
20
```

Newlib does not access the FPU (expected):
```
cd /opt/zephyr-sdk-0.16.3/sparc-zephyr-elf/
./bin/sparc-zephyr-elf-objdump -d sparc-zephyr-elf/lib/leon/soft/libm.a | grep %fsr | wc -l
0
```

Picolibc 1.8.4 and later contains a fix which resolves this: https://github.com/zephyrproject-rtos/picolibc/commit/c014f803406d260b152007f1642c36a3fd04ee26

It is expected that pull request #707 will close this issue.

tbr-tt, 2023-10-03:
@keith-packard  FYI

stephanosio, 2023-11-14:
Fixed by https://github.com/zephyrproject-rtos/sdk-ng/pull/707.

Please re-open if this issue still persists in the upcoming SDK 0.16.4 release.

closed 2023-11-14 by stephanosio