zephyrproject-rtos/sdk-ng #720: does SDK support riscv vector extension

yf13, 2023-11-26:
I am wondering can our SDK supports target with riscv64 vector extension v1.0?

closed 2024-01-25 by stephanosio

haijun20222, 2025-03-14:
Hello, as you described above, I am using the latest official riscv64-zephyr-elf toolchain to compile riscv64 CPUs with vector support and encountered compilation issues. I need to contribute new code to Zephyr, but there are problems compiling riscv64 with riscv vector support. Currently, I have the following questions: 
1. Which RISC-V CPU types is the riscv64-zephyr-elf toolchain applicable to, and what is its application scope? In other words, what is the purpose of this toolchain's design? 
2. Will the v extension be continued for riscv64 in the future? If not, do I need to provide my own toolchain support when contributing code?

haijun20222, 2025-03-14:
Hello, as you described above, I am using the latest official riscv64-zephyr-elf toolchain to compile riscv64 CPUs with vector support and encountered compilation issues. I need to contribute new code to Zephyr, but there are problems compiling riscv64 with riscv vector support. Currently, I have the following questions:

Which RISC-V CPU types is the riscv64-zephyr-elf toolchain applicable to, and what is its application scope? In other words, what is the purpose of this toolchain's design?
Will the v extension be continued for riscv64 in the future? If not, do I need to provide my own toolchain support when contributing code?