zephyrproject-rtos/sdk-ng #724: Individual toolchain installation should be ignored for full_sdk

asemjonovs, 2023-12-05:
Running `setup.sh -t all` with a Full SDK downloads and installs all the toolchains again.  This should be ignored or provide a warning when using the Full SDK.

yperess, 2023-12-05:
For more information on this, we have a build environment that doesn't have `wget`. We would also need the `check_command wget 91` line moved to just before it's actually used. I think that if we move it into the loop `for toolchain in ${inst_toolchains[@]}; do` then it would work.

stephanosio, 2024-09-22:
> Running setup.sh -t all with a Full SDK downloads and installs all the toolchains again. This should be ignored or provide a warning when using the Full SDK.

@asemjonovs I am unable to reproduce this.

```
stephanos@devworx2 ~/Downloads/zephyr-sdk-0.16.8                                                                                                                                                       [06:47:54]
$ ls -l
total 42884
drwxr-xr-x 9 stephanos stephanos     4096 May 29 22:08 aarch64-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 30 01:14 arc-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 22:14 arc64-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 30 00:32 arm-zephyr-eabi
drwxr-xr-x 3 stephanos stephanos     4096 May 29 20:26 cmake
drwxr-xr-x 8 stephanos stephanos     4096 May 29 22:38 microblazeel-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 22:17 mips-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 22:30 nios2-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 30 00:16 riscv64-zephyr-elf
-rw-r--r-- 1 stephanos stephanos      612 May 30 17:39 sdk_toolchains
-rw-r--r-- 1 stephanos stephanos        7 May 30 17:39 sdk_version
-rwxr-xr-x 1 stephanos stephanos     5570 May 30 17:39 setup.sh
drwxr-xr-x 9 stephanos stephanos     4096 May 29 23:08 sparc-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:27 x86_64-zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:20 xtensa-dc233c_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:13 xtensa-espressif_esp32_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:32 xtensa-espressif_esp32s2_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:12 xtensa-espressif_esp32s3_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:13 xtensa-intel_ace15_mtpm_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:42 xtensa-intel_tgl_adsp_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:47 xtensa-mtk_mt8195_adsp_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:58 xtensa-nxp_imx8m_adsp_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:41 xtensa-nxp_imx8ulp_adsp_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:46 xtensa-nxp_imx_adsp_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:50 xtensa-nxp_rt500_adsp_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:46 xtensa-nxp_rt600_adsp_zephyr-elf
drwxr-xr-x 9 stephanos stephanos     4096 May 29 21:46 xtensa-sample_controller_zephyr-elf
-rwxr-xr-x 1 stephanos stephanos 43796074 May 29 21:23 zephyr-sdk-x86_64-hosttools-standalone-0.9.sh

stephanos@devworx2 ~/Downloads/zephyr-sdk-0.16.8                                                                                                                                                       [06:48:25]
$ ./setup.sh -t all
Zephyr SDK 0.16.8 Setup

All done.
```

The current toolchain installation logic already checks if a toolchain directory exists before attempting to download and install the toolchain:

https://github.com/zephyrproject-rtos/sdk-ng/blob/20344effef69890eb2a3c39b601e0a05dc183692/scripts/template_setup_posix#L210-L213

stephanosio, 2024-09-22:
> For more information on this, we have a build environment that doesn't have `wget`. We would also need the `check_command wget 91` line moved to just before it's actually used. I think that if we move it into the loop `for toolchain in ${inst_toolchains[@]}; do` then it would work.

I do not understand what you mean by this. `check_command wget 91` is already invoked before `wget` is invoked for the first time:

https://github.com/zephyrproject-rtos/sdk-ng/blob/20344effef69890eb2a3c39b601e0a05dc183692/scripts/template_setup_posix#L197

The first and the only invocation of `wget` in the script:

https://github.com/zephyrproject-rtos/sdk-ng/blob/20344effef69890eb2a3c39b601e0a05dc183692/scripts/template_setup_posix#L217-L218

nashif, 2025-05-24:
is this still an issue?