zephyrproject-rtos/sdk-ng #726: `setup.sh` cannot find applications on PATH

waffle87, 2023-12-16:
When running the `setup.sh` script as [documented here](https://docs.zephyrproject.org/3.2.0/develop/getting_started/index.html#install-zephyr-sdk), I get the error
```
$ ./setup.sh
Zephyr SDK 0.16.4 Setup

Zephyr SDK setup requires 'cmake' to be installed and available in the PATH.
Please install 'cmake and run this script again.
```
I have cmake 3.28.0 installed, and it is in fact in my PATH
```
$ echo $PATH
/home/jack/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin

$ which cmake
/usr/bin/cmake
```
If I comment out the lines
```shell
# Check dependencies
check_command cmake 90
check_command wget 91
```
the script reaches
```
Install host tools [y/n]? y
Register Zephyr SDK CMake package [y/n]? y

Installing host tools ...
Zephyr Yocto Toolchain SDK installer version 0.9
================================================
You are about to install the SDK to "/home/jack/zephyr-sdk-0.15.0". Proceed [Y/n]? Y
Extracting SDK....................xz: (stdin): Unexpected end of input
tar: Child returned status 1
tar: Error is not recoverable: exiting now
ERROR: Host tools installation failed
```
Which I'm unsure if is related to the "missing" commands. It's worth noting `wget` is also not found if I only comment out the `check_command cmake 90` line. Operating system is Linux with ZSH.

I also tried running the `zephyr-sdk-x86_64-hosttools-standalone-0.9.sh` script, which finishes with
```
$ ./zephyr-sdk-x86_64-hosttools-standalone-0.9.sh
Error: The SDK needs a python installed
```

stephanosio, 2024-01-25:
This seems to be an issue with your local environment.

Try creating `test.sh` with the following content and see if it works:

```
#!/usr/bin/env bash
which cmake
```

waffle87, 2024-01-28:
```sh
$ ./test.sh
./test.sh: line 2: which: command not found
```
Ah, this does seem to be local to me then, I'll see if I can sort it out. Apologies for the noise.

closed 2024-01-28 by waffle87