zephyrproject-rtos/sdk-ng #741: pokysdk provides an older Verison of GLIBC than required by system

HannesD1, 2024-02-28:
Hi there,

i have installed Zepyhr and the SDK according to the zephyr website.
When i tried to build an example i go the following error message.
`/home/hannes/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: /home/hannes/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/lib/libc.so.6: version `GLIBC_2.38' not found (required by /lib64/libstdc++.so.6)
`
Changing the symlink in pokysdk to my local glibc installation in /lib4 did not helped.
Shouldnt pokysdk provide the full environment to build an application since its using my local /lib64/libstdc++ or did i got this wrong ?
I think this is somehow similar to this issue https://github.com/zephyrproject-rtos/sdk-ng/issues/584 but just inverted.

Zephyr SDK Version 16.5
I am running Fedora 39

stephanosio, 2024-02-28:
You might have relocated the Zephyr SDK installation after running `setup.sh`. Please run `setup.sh` again to fix RPATHs.

HannesD1, 2024-02-29:
Running setup.sh did not fix the problem. The installation is loacted in ~/zephyr-sdk-0.16.5 As described on the zephyr page

HannesD1, 2024-07-14:
I am still encountering this problem. I was able to fix it by creating a hard reference to use the installed DTC application from the computer and not the one from Poky. But this obviously doesn't feel right. Is it the intended behavior that west should use the Poky DTC or the DTC installed on the computer? Since we have to install DTC anyway according to the installation guide, the DTC provided by Poky seems somewhat redundant oder does this one serves a special purpose ?

HannesD1, 2024-07-15:
Due to multiple error based on that wrong GLIBC version I have just created a Symlink to my system /lib64 dir in the pokysdk folder. This seems to work for me

closed 2024-09-19 by stephanosio