zephyrproject-rtos/sdk-ng #751: v0.16.6 cannot link samples/cpp/hello_world out-of-the-box

cfriedt, 2024-05-01:
Just test-driving the v0.16.6 SDK, and it looks like `samples/cpp/hello_world` does not build and run OOTB.

Moving from v0.16.5-1 to 0.16.6, I needed to additionally enable the following in order for the app to link.

```shell
CONFIG_COMMON_LIBC_THRD=y
CONFIG_POSIX_API=y
CONFIG_DYNAMIC_THREAD=y
CONFIG_THREAD_STACK_INFO=y
```

One possible solution for the above would be to create weak versions of the [C11 <thread.h> functions](https://en.cppreference.com/w/c/thread) in the toolchain, such that linking to an external implementation will provided the necessary functionality (when required).

Since `samples/cpp/hello_world` does not require C++11 threading, this seems like it would be a good choice.

Beyond that, a couple of additional platforms are now facing issues

**qemu_riscv64** and **qemu_riscv32**

```shell
[142/143] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: riscv64
*** Booting Zephyr OS build v3.6.0-3273-g40a8022e4a14 ***
abort()
```

```shell
[142/143] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: riscv32
*** Booting Zephyr OS build v3.6.0-3273-g40a8022e4a14 ***
abort()
```

This is likely due to something like stack unwinding, but it will take additional effort to diagnose / remedy.

stephanosio, 2024-05-05:
This effectively renders the SDK 0.16.6 release unusable until the C11 threads stub functions are added to the Zephyr `main` branch.

**Also, this is a breaking change that creates an incompatibility between the SDK 0.16.6 release and any Zephyr releases prior to the upcoming 3.7.0, which we do not want at this point.**

This should have been caught earlier and addressed in a more self-contained way within the SDK (e.g. providing weak C11 threads stub functions within the libgcc itself); unfortunately, the SDK CI does not build and run the C++ tests and samples and this was not caught -- I am going to fix that by adding more tests to be run in the SDK CI.

At this point, I will need to mark the 0.16.6 a bad release and release 0.16.7 without the libstdc++ C11 gthread patch. The libstdc++ patch will need to be reworked to handle the transition more gracefully (as mentioned above, by providing weak C11 threads stubs).

stephanosio, 2024-05-08:
Marking it as an enhancement instead of a bug since 0.16.7 has been released without the C11 threads-based gthread libgcc patch.

markaj-nordic, 2024-05-10:
FYI:
After switching to the 0.16.6, without any tweaks, our C++ Matter applications won't build (due to the linkage issue).
I will collect build logs and post it here.
Any ETA for 0.16.7 release? I see the tag is already there.

seyednasermoravej, 2024-06-01:
Hi, I'm facing the same issue as @markaj-nordic mentioned.
Any progress of instructions to solve this issue?
Tnx

cfriedt, 2024-06-08:
@seyednasermoravej  use 0.16.8