zephyrproject-rtos/sdk-ng #771: setup.sh wget invocation doesn't work with wget2 in newer distros: "Unknown option 'show-progress'"

karlp, 2024-06-21:
The wget --show-progress line here
https://github.com/zephyrproject-rtos/sdk-ng/blob/main/scripts/template_setup_posix#L218
doesn't work with the wget2 installed as wget on, for instance, fedora40.

I'm not sure what the right solution here is :)

using `--progress=bar:force` instead of `--show-progress`  _appears_ to be both forwards and backwards compatible, but I don't have extensive tests, just reading the man pages for both versions.

captcha1, 2024-08-09:
I get this error on Fedora 40 as well.

Maybe should be a separate bug report, but I also notice it does "wget", then "tar" and removes the intermediate file, whereas it's faster and uses less disk space to do : 

wget -O - "${toolchain_uri}" | tar xJ

stephanosio, 2024-09-19:
After some investigation, it looks like Fedora made a very bad decision to replace `wget` with `wget2`, which is not intended to be a drop-in replacement and "just not ready."

Some noteworthy comments from the upstream wget developers:

> [It has recently come to our attention that Fedora 40 made the decision to ship GNU Wget2 as GNU Wget. While both the projects are maintained by the same developers and Wget2 is indeed supposed to be the natural successor to Wget, it is just not ready. Upstream was NOT asked for its opinion on this matter. If we had been asked about this, we'd have stated our concerns and said that it's a bad idea.](https://gitlab.com/gnuwget/wget2/-/issues/661)

> [We've never claimed Wget2 is ready to be used as Wget. It is a binary called Wget2 and we intend for it to be used in parallel with Wget, atleast for the time being.](https://gitlab.com/gnuwget/wget2/-/issues/661)

Fedora change proposal for wget2: https://discussion.fedoraproject.org/t/f40-change-proposal-wget2-as-wget/96422/6

At this point, it is up to the Fedora team to revert this change, not the downstream projects that expect wget to be wget -- for a workaround, please refer to [the relevant threads on the Fedora forum](https://forums.fedoraforum.org/showthread.php?332571-F40-wget2-breaks-wget-compatibility&s=82fd82821300be0a09f7971e05804a21&p=1884011#post1884011).

closed 2024-09-19 by stephanosio

stephanosio, 2024-11-21:
Also note that RHEL 10 will be shipping with `wget` as "real wget," not wget2; only confirming that this was a mistake.

https://src.fedoraproject.org/rpms/wget2/c/a05eb637c21e1fa06926ef7e963518572f8cdbad

i509VCB, 2025-01-30:
A new note on this, wget2 has added a compatability option for `--show-progress`, so eventually new Fedora versions will not have this issue.

https://gitlab.com/gnuwget/wget2/-/merge_requests/558 (Manually merged)