zephyrproject-rtos/sdk-ng #824: Ubuntu 24 LTS: Scripting in the "Python" language is not supported in this copy of GDB

SeppoTakalo, 2024-10-16:
```
❯ arm-zephyr-eabi-gdb
GNU gdb (Zephyr SDK 0.16.8) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arm-zephyr-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) load_dashboard 
~/.gdbinit.dashboard:2294: Error in sourced command file:
Scripting in the "Python" language is not supported in this copy of GDB.

quit
❯ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble

```

Why is the Python disabled on Ubuntu 24.04? Is this intentional?
It used to work previously, but I'm unsure if this has anything to do with the distro upgrade.

stephanosio, 2024-10-16:
Please use `gdb-py` (e.g. `arm-zephyr-eabi-gdb`) if you need Python scripting.

SeppoTakalo, 2024-10-16:
Ah.. I see
```
❯ arm-zephyr-eabi-gdb-py
arm-zephyr-eabi-gdb-py: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
```

So this binary ships with dynamically linked to libpython but not on a version that ships with current Ubuntu LTS.

stephanosio, 2024-10-16:
> So this binary ships with dynamically linked to libpython but not on a version that ships with current Ubuntu LTS.

gdb-py is linked against the minimum supported Python version on the Zephyr side. From the upcoming SDK 0.17.0 (and 0.16.9 in case of LTS) release, it will be linked against Python 3.10 since Zephyr now requires Python 3.10 and above.

If you are on Ubuntu, you can either install the required version of Python from source or use a fairly well-maintained PPA like [deadsnakes](https://launchpad.net/~deadsnakes/+archive/ubuntu/ppa).