zephyrproject-rtos/sdk-ng #833: Make Picolibc the default libc and Newlib an optional component for SDK GNU toolchain

stephanosio, 2024-11-01:
Since Picolibc is now the default libc for Zephyr serving as a complete replacement of Newlib and only niche applications are using Newlib, there is no good reason to ship Newlib in the SDK GNU toolchain distribution by default in spite of its significant download size and disk space requirements.

With the above in mind, this proposes to rearrange the SDK GNU toolchain distribution such that:

1. Picolibc is the default libc under the toolchain prefix (i.e. `(TOOLCHAIN_PREFIX)/(TARGET)/lib/lib{c,m}.a` and the headers under `(PREFIX)/(TARGET)/include` will be from Picolibc).
2. Newlib is placed under its own prefix, `(TOOLCHAIN_PREFIX)/newlib`, and distributed as a separate "overlay" archive, `toolchain_overlay_(TARGET).{tar.xz,7z}`, that can be downloaded optionally.

stephanosio, 2024-11-01:
cc @keith-packard

keith-packard, 2024-11-01:
Yup, this might take a bit of re-arranging inside crosstool-ng, but we've already demonstrated we can make newlib work this way for newlib-nano.

closed 2025-05-26 by nashif