zephyrproject-rtos/sdk-ng #835: Centos 7 build errors

dpetrisko, 2024-11-08:
Hello! I am getting two errors when using west sdk install on Centos 7. I'm manually installing with https://docs.zephyrproject.org/latest/develop/toolchains/zephyr_sdk.html#toolchain-zephyr-sdk-install to get around the issues, so not a showstopper.  My question is for the zephyr/west team. Is centos 7 considered deprecated or would you appreciate a patch? These seem like fairly trivial fixes, but I don't want to but OS-specific switches if they're not wanted. Thank you!

```
(.venv) [petrisko@kk9 zephyr]$ west sdk install -b $PWD
Found '/mnt/users/ssd0/homes/petrisko/research/black-parrot-sim/black-parrot-sdk/zephyr/zephyr/SDK_VERSION', installing version 0.17.0.
Fetching Zephyr SDK list...
Fetching sha256...
Downloading https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.17.0/zephyr-sdk-0.17.0_linux-x86_64_minimal.tar.xz...
 44517504/44517504
Downloaded: /mnt/users/ssd0/homes/petrisko/research/black-parrot-sim/black-parrot-sdk/zephyr/tmpl7ynxrus/zephyr-sdk-0.17.0_linux-x86_64_minimal.tar.xz
Extract: /mnt/users/ssd0/homes/petrisko/research/black-parrot-sim/black-parrot-sdk/zephyr/tmpl7ynxrus/zephyr-sdk-0.17.0_linux-x86_64_minimal.tar.xz
INFO patool: Extracting /mnt/users/ssd0/homes/petrisko/research/black-parrot-sim/black-parrot-sdk/zephyr/tmpl7ynxrus/zephyr-sdk-0.17.0_linux-x86_64_minimal.tar.xz ...
/usr/bin/file: unrecognized option '--no-sandbox'
Usage: file [-bchikLlNnprsvz0] [--apple] [--mime-encoding] [--mime-type]
            [-e testname] [-F separator] [-f namefile] [-m magicfiles] file ...
       file -C [-m magicfiles]
       file [--help]
```

```
All done.
Zephyr SDK 0.17.0 Setup

Installing 'aarch64-zephyr-elf' toolchain ...
wget: unrecognized option '--show-progress'
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
ERROR: Toolchain download failed
FATAL ERROR: command "/mnt/users/ssd0/homes/petrisko/research/black-parrot-sim/black-parrot-sdk/zephyr/zephyr-sdk-0.17.0/setup.sh -t all -h" failed
```

dkalowsk, 2024-11-08:
@stephanosio this may be more your area

stephanosio, 2024-11-10:
RHEL 7 has reached end of maintenance and we no longer support it. Also note that [the minimum glibc version required by Zephyr SDK is 2.28](https://github.com/zephyrproject-rtos/sdk-ng/issues/764), which is going to be a problem for RHEL/CentOS 7 with 2.17.

closed 2024-11-10 by stephanosio

stephanosio, 2024-11-10:
FWIW, I have just tested RHEL 8.10 and, as of Zephyr SDK 0.17.0, it seems to be working fine.